amov-lab/Prometheus #55: ./compile_gazebo_simulator.sh 运行出错

WANGshouming4937, 2020-08-30:
[ 33%] Built target trajectory_msgs_generate_messages_lisp
[ 33%] Built target trajectory_msgs_generate_messages_nodejs
[ 33%] Built target prometheus_msgs_generate_messages_lisp
[ 50%] Building CXX object CMakeFiles/realsense_gazebo_plugin.dir/gazebo_plugin/RealSensePlugin.cpp.o
In file included from /home/wsh/Prometheus/Simulator/gazebo_simulator/include/RealSensePlugin.h:20:0,
                 from /home/wsh/Prometheus/Simulator/gazebo_simulator/gazebo_plugin/RealSensePlugin.cpp:17:
/usr/include/gazebo-9/gazebo/common/Plugin.hh:33:10: fatal error: sdf/sdf.hh: No such file or directory
 #include <sdf/sdf.hh>
          ^~~~~~~~~~~~
compilation terminated.
CMakeFiles/realsense_gazebo_plugin.dir/build.make:62: recipe for target 'CMakeFiles/realsense_gazebo_plugin.dir/gazebo_plugin/RealSensePlugin.cpp.o' failed
make[2]: *** [CMakeFiles/realsense_gazebo_plugin.dir/gazebo_plugin/RealSensePlugin.cpp.o] Error 1
CMakeFiles/Makefile2:2287: recipe for target 'CMakeFiles/realsense_gazebo_plugin.dir/all' failed
make[1]: *** [CMakeFiles/realsense_gazebo_plugin.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1 -l1" failed
执行 make px4_sitl gazebo 报错
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/wsh/Prometheus/Simulator/gazebo_simulator/models/MultiUAV"
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/wsh/Prometheus/Simulator/gazebo_simulator/models/race_track"
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/wsh/Prometheus/Simulator/gazebo_simulator/models/texture"
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/wsh/Prometheus/Simulator/gazebo_simulator/models/texture_amov"
INFO  [ecl/EKF] 6052000: EKF commencing GPS fusion
将下面的环境变量注释后，运行make px4_sitl gazebo正常
export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:${your prometheus path}/Prometheus/devel/lib
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:${your prometheus path}/Prometheus/Simulator/gazebo_simulator/models
source ${your px4 path}/Firmware_v110/Tools/setup_gazebo.bash ${your px4 path}/Firmware_v110 ${your px4 path}/Firmware_v110/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:${your px4 path}/Firmware_v110
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:${your px4 path}/Firmware_v110/Tools/sitl_gazebo

WANGshouming4937, 2020-08-31:
今天又试了下，应该安装的gazebo有问题，卸载之后重新安装好了

closed 2021-01-09 by potato77

imposin, 2021-08-25:
运行cd Prometheus
./compile_gazebo.sh
的时候出现一样的错误  重装gazebo没有好