amov-lab/Prometheus #58: autonomous_landing碰到的一些问题

L3Y1Q2, 2020-09-02:
@potato77 ，看了大佬的关于无人机降落小车的论文，希望有机会能交流交流。
**我用的固件版本是PX4 master版本，所以不确定碰到的问题是不是因为固件的问题。**
问题1：
[image]
有if (Command_Last.Mode != prometheus_msgs::ControlCommand::Land)时候，无人机总是悬停在标识的distance_thres高度处不会下降，调试发现是因为_Reference_State.Move_mode会变为位置模式，去掉这个if判断条件就没有这个现象了。
问题2：
[image]
无人机降落在小车上无法disarm，因为if(distance_to_setpoint < distance_thres)这个条件会一直满足，不会执行到else if(abs(landpad_det.pos_body_enu_frame[2]) < 0.3)，同时在px4_pos_controller.cpp文件中的 if(abs(_DroneState.position[2] - Takeoff_position[2]) < Disarm_height)因为无人机不是在小车上起飞的，所以也无法disarm。
问题3：
CPID NOT SUPPOORT TRAJECTORY TRACKING，这里指的是trajectory 而不是path ，所以cpid不能支持，不知道我理解的对不对？

potato77, 2020-09-02:
问题太多了，这些问题都涉及代码二次开发了，建议报名在线课程学习

L3Y1Q2, 2020-09-03:
> 问题太多了，这些问题都涉及代码二次开发了，建议报名在线课程学习

谢谢回复，大佬们写的代码逻辑很清晰，二次开发还是很方便的。上面这些倒也不是啥问题，修改了之后仿真基本上能达到想要的效果，只不过因为没有代码运行的原始效果对比，有点不确定，哈哈。

potato77, 2021-01-09:
降落这块有更新，可更新代码尝试一下

closed 2021-01-09 by potato77