amov-lab/Prometheus #61: Fast Planner不规划问题

SaligiaR, 2020-09-06:
运行`roslaunch prometheus_gazebo sitl_fast_planning_3dlidar.launch`，按照wiki教程操作，无法进行规划，rviz和gazebo中无人机不规划，运行`rostopic echo /prometheus/control_command`，没有相关信息滚动，全局规划是有对应控制信息滚动显示的，planning_mission界面显示如图，一直在等待轨迹，运行`rostopic echo /prometheus/planning/goal`是有目标点随topic发出的。同样地，RGBD也有相应的问题，RGBD还存在坐标系问题：

> Error:   TF_SELF_TRANSFORM: Ignoring transform from authority "/rtabmap/rtabmap" with frame_id and child_frame_id  "world" because they are the same
[image]

Amit10311, 2020-10-07:
Dear @SaligiaR, I would like to know is your controller is workers and are you using hardware or only in a simulation?

GanAH, 2021-04-01:
> 运行`roslaunch prometheus_gazebo sitl_fast_planning_3dlidar.launch`，按照wiki教程操作，无法进行规划，rviz和gazebo中无人机不规划，运行`rostopic echo [...]

你好，请问你的无人机是用键盘控制起飞的吗？我的无人机无法连接，不知道大神怎么做的呢？

SaligiaR, 2021-04-12:
> Dear @SaligiaR, I would like to know is your controller is workers and are you using hardware or only in a simulation?

In this work, I don't use px4, just use gazebo and rviz. And I can control it in gazebo.

SaligiaR, 2021-04-12:
@GanAHE 我是使用rviz指定航点进行规划，同时我也成功用手柄控制过。

SaligiaR, 2021-04-12:
半年过去了没有解答，我也早就从事其他方面的研究了，因此我决定关闭这个问题

closed 2021-04-12 by SaligiaR