amov-lab/Prometheus #64: 仿真迁移至NVIDA Nano  /mavros/state 订阅消息与发布消息不一致问题

YANG-DEMIN, 2020-09-13:
硬件：NVIDA nano 运行阿木预装系统 Linux version 4.9.140-tegra，如图一所示
问题描述：
1. 可以正常运行 **roslaunch mavros px4.launch fcu_url:=”/dev/ttyUSB0:921600**
2. **rostopic echo /mavros/imu/data 数据正常；  rostopic echo /mavros/state 数据异常，报错如下**
[image]
[image]
3. 同样，在运行node 时，也会报错

473867143, 2020-09-14:
请参考论坛：https://bbs.amovlab.com/forum.php?mod=viewthread&tid=26810&extra=page%3D1
最后重新编译工作区间

YANG-DEMIN, 2020-09-14:
感谢解答！
按照您说的方法去做了，还是不行，mavros已经更新到1.3。
重启之后，使用rostopic echo /mavros/…… 等命令时，终端无反映。
希望再次解答。
个人猜测是不是本身NANO环境配置存在问题，我想给nano重新安装系统或者配置环境，有没有什么建议？
[image]
[image]

473867143, 2020-09-15:
多谢提问，
1、我也尝试用NANO通过串口连接到飞控中，发现也是`echo`不到`/mavros/state` ，这一点很奇怪，没找到为什么打印不出来这个消息。但是可以明确的一点是，如果你上面截图中的报错原因，那肯定是mavros版本不一致导致哈希值出现的问题，你的报错有两种可能，第一种是mavros及其插件安装的版本不一致，这种情况可以排除，我这边也是同样的版本；第二种是你的工作区间需要重新编译，删除之前编译文件`build和devel`，重新catkin_make，不然也会出现一样的问题，应为他是上个mavros版本生成的编译文件。
2、即使没有`echo 出来 /mavros/state`的状态，也可以判断NANO是否连接上飞控，第一种板载计算机上查看`roslaunch mavros px4.launch fcu_url:="/dev/ttyTHS1:921600"`
[image]
已有心跳包，并显示了飞控软件的版本，操作系统的版本，等等；或者使用`rostopic echo -n1 /diagnostics`
[image]
也可以看出mavros与飞控已经正常连接。
还有一种方式是通过地面站`mavlink console`查看,输入`mavlink status`
[image]
可以看出来，第一个是usb连接到NANO之上，第二个是telem1连接到qgc上，第三个就是telem2连接到了NANO。
3、我在NANO上运行激光雷达室内见图定位的脚本，

[image]

可以看到飞机的状态栏出，是显示已连接状态。这时候`rostopic info /mavros/state` 可以看出是有订阅到飞控的状态并且也能正常的显示。
[image]

综上的话，不能直接打印显示mavros/state的确是有问题，但是貌似不影响数据流，你提的问题也很好，我们在查找一下为什么不能直接打印显示。

YANG-DEMIN, 2020-09-15:
非常感谢回答！问题已经解决，把build和devel文件删除然后重新编译就好了。
现在的情况是个别消息可以打印，比如/mavros/imu/data，/mavros/battery。但是绝大多数消息还是不能打印，比如state，/imu/mag。
不清楚问题出在哪儿了。

closed 2020-09-15 by YANG-DEMIN

YANG-DEMIN, 2021-01-10:
在新的纯净系统上出现同样的问题，查找资料后终于发现，果然还是mavros版本的问题。
解决方法就是将地面计算机和机载计算机的mavros卸载后再安装最新的版本（如果发现有一个已经是最高版本了，可不必对其操作）。
sudo apt-get remove ros-melodic-mavros*
然后使用catkin_make clean (或者catkin clean) 将之前的编译删除
最后重新安装 
sudo apt-get install ros-melodic-mavros ros-melodic-mavros-extras