amov-lab/Prometheus #66: 关于autonomous landing的问题

bamboosdu, 2020-09-29:
Hi， 我看到在演示视频中无人机能根据marker去做自主降落
其中一个关键部分是如何去识别marker，并且定位marker的坐标，我在源码中没有看到，不知道release了没有

potato77, 2020-09-29:
视觉识别部分吗？有的啊，在object_detection里面，你看launch启动了哪个节点就知道了

bamboosdu, 2020-09-30:
非常感谢您第一时间的回复，我今天看了下，确实在object_detection里面
对于检测逻辑我理解如果没错的话：先检测中心的小marker，再去检测中心的大marker，再去检测四周的4个marker
我觉得是不是可以做这样的改进，基于大marker的边长相对较大，所以先检测大marker会更准确一些，在检测不到中心的大marker后再去检测小marker。如果我的思路有问题烦请纠正。
另外一个问题是，我发现四周的小marker并没有根据它们相对应的位置去做偏移，偏移它们的中心点到大、小marker的中心点
如下图所示
[image]
[image]

jario-jin, 2020-10-01:
我们的思路是这样的：
优先检查小marker是为了近距离时更精确的定位信息，如果检测不到小marker，才会检测大marker（在远距离时保持定位信息），周围4给marker是在近距离时，无人机控制精度不够，导致远离中心点，周围的marker保持鲁棒性，不会脱离目标。

bamboosdu, 2020-10-20:
`           if (-1 != t)
            {
                // 解算位置的平均值
                // cout << marker_count << endl;
                cv::Point3f A1_Position_OcInW(0,0,0);
                double A1_yaw = 0.0;
                A1_Position_OcInW = A1_Sum_Position_OcInW / marker_count;
                A1_yaw = A1_Sum_yaw / marker_count;

                // 将解算后的位置发给控制端
                pose_now.header.stamp = ros::Time::now();
                pose_now.detected = true;
                pose_now.frame = 0;
                pose_now.position[0] = tx;
                pose_now.position[1] = ty;
                pose_now.position[2] = tz;
                pose_now.sight_angle[0] = atan(tx / tz);
                pose_now.sight_angle[1] = atan(ty / tz);
                pose_now.yaw_error = A1_yaw;

                last_x = pose_now.position[0];
                last_y = pose_now.position[1];
                last_z = pose_now.position[2];
                last_yaw = pose_now.yaw_error;
            }
`

closed 2020-11-24 by bamboosdu