amov-lab/Prometheus #74: 关于compile_planning.sh编译的错误

lanqx1003, 2020-10-28:
编译compile_planning.sh的时候出现下面的错误
[image]
[image]
这个错误是和前面出现的这个warning有关吗？
编译所有后虽然有这个错误但是能正常运行APF局部规划的仿真程序
/home/lanqx/Prometheus/Modules/planning/FastPlanner/plan_env/ThirdParty/sdf_tools/src/sdf_tools/sdf_builder.cpp: In member function ‘bool sdf_tools::SDF_Builder::BuildInternalPlanningScene()’:
/home/lanqx/Prometheus/Modules/planning/FastPlanner/plan_env/ThirdParty/sdf_tools/src/sdf_tools/sdf_builder.cpp:87:134: error: no matching function for call to ‘planning_scene::PlanningScene::PlanningScene(boost::shared_ptr<urdf::Model>&, boost::shared_ptr<srdf::Model>&)’
 ene::PlanningScene>(new planning_scene::PlanningScene(urdf_model, srdf_model));
                                                                             ^
In file included from /home/lanqx/Prometheus/Modules/planning/FastPlanner/plan_env/ThirdParty/sdf_tools/include/sdf_tools/sdf_builder.hpp:9:0,
                 from /home/lanqx/Prometheus/Modules/planning/FastPlanner/plan_env/ThirdParty/sdf_tools/src/sdf_tools/sdf_builder.cpp:12:
/opt/ros/melodic/include/moveit/planning_scene/planning_scene.h:966:3: note: candidate: planning_scene::PlanningScene::PlanningScene(const PlanningSceneConstPtr&)
   PlanningScene(const PlanningSceneConstPtr& parent);
   ^~~~~~~~~~~~~
/opt/ros/melodic/include/moveit/planning_scene/planning_scene.h:966:3: note:   candidate expects 1 argument, 2 provided
/opt/ros/melodic/include/moveit/planning_scene/planning_scene.h:96:3: note: candidate: planning_scene::PlanningScene::PlanningScene(const ModelInterfaceSharedPtr&, const ModelConstSharedPtr&, const WorldPtr&)
   PlanningScene(
   ^~~~~~~~~~~~~
/opt/ros/melodic/include/moveit/planning_scene/planning_scene.h:96:3: note:   no known conversion for argument 1 from ‘boost::shared_ptr<urdf::Model>’ to ‘const ModelInterfaceSharedPtr& {aka const std::shared_ptr<urdf::ModelInterface>&}’
/opt/ros/melodic/include/moveit/planning_scene/planning_scene.h:90:3: note: candidate: planning_scene::PlanningScene::PlanningScene(const RobotModelConstPtr&, const WorldPtr&)
   PlanningScene(
   ^~~~~~~~~~~~~
/opt/ros/melodic/include/moveit/planning_scene/planning_scene.h:90:3: note:   no known conversion for argument 1 from ‘boost::shared_ptr<urdf::Model>’ to ‘const RobotModelConstPtr& {aka const std::shared_ptr<const moveit::core::RobotModel>&}’
FastPlanner/plan_env/ThirdParty/sdf_tools/CMakeFiles/sdf_tools.dir/build.make:158: recipe for target 'FastPlanner/plan_env/ThirdParty/sdf_tools/CMakeFiles/sdf_tools.dir/src/sdf_tools/sdf_builder.cpp.o' failed
make[2]: *** [FastPlanner/plan_env/ThirdParty/sdf_tools/CMakeFiles/sdf_tools.dir/src/sdf_tools/sdf_builder.cpp.o] Error 1
CMakeFiles/Makefile2:4738: recipe for target 'FastPlanner/plan_env/ThirdParty/sdf_tools/CMakeFiles/sdf_tools.dir/all' failed
make[1]: *** [FastPlanner/plan_env/ThirdParty/sdf_tools/CMakeFiles/sdf_tools.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

XunNie, 2020-11-13:
修改Modules/planning/FastPlanner/plan_env/ThirdParty/sdf_tools/src/sdf_tools/sdf_builder.cpp 第80行-89行

diff --git a/Modules/planning/FastPlanner/plan_env/ThirdParty/sdf_tools/src/sdf_tools/sdf_builder.cpp b/Modules/planning/FastPlanner/plan_env/ThirdParty/sdf_tools/src/sdf_tools/sdf_builder.cpp
index d585497..a217218 100644
--- a/Modules/planning/FastPlanner/plan_env/ThirdParty/sdf_tools/src/sdf_tools/sdf_builder.cpp
+++ b/Modules/planning/FastPlanner/plan_env/ThirdParty/sdf_tools/src/sdf_tools/sdf_builder.cpp
@@ -77,10 +77,12 @@ namespace sdf_tools
     {
         /* Builds a planning scene from XML string urdf and srdf descriptions */
         // Make the URDF model
-        boost::shared_ptr<urdf::Model> urdf_model(new urdf::Model());
+        //boost::shared_ptr<urdf::Model> urdf_model(new urdf::Model());
+        urdf::ModelSharedPtr urdf_model(new urdf::Model());
         urdf_model->initString(GenerateSDFComputeBotURDFString());
         // Make the SRDF model
-        boost::shared_ptr<srdf::Model> srdf_model(new srdf::Model());
+        //boost::shared_ptr<srdf::Model> srdf_model(new srdf::Model());
+        srdf::ModelSharedPtr srdf_model(new srdf::Model());
         srdf_model->initString(*urdf_model, GenerateSDFComputeBotSRDFString());
         // Make the planning scene
         planning_scene_ptr_.reset();

XunNie, 2020-11-13:
Modules/planning/FastPlanner/plan_env/ThirdParty/sdf_tools/src/sdf_tools/sdf_builder.cpp 第80行左右, 修改参考:
` 80         //boost::shared_ptr<urdf::Model> urdf_model(new urdf::Model());
 81         urdf::ModelSharedPtr urdf_model(new urdf::Model());
 82         urdf_model->initString(GenerateSDFComputeBotURDFString());
 83         // Make the SRDF model
 84         //boost::shared_ptr<srdf::Model> srdf_model(new srdf::Model());
 85         srdf::ModelSharedPtr srdf_model(new srdf::Model());
 86         srdf_model->initString(*urdf_model, GenerateSDFComputeBotSRDFString());
 87         // Make the planning scene
 88         planning_scene_ptr_.reset();
 89         planning_scene_ptr_ = std::shared_ptr<planning_scene::PlanningScene>(new planning_scene::PlanningS    cene(urdf_model, srdf_model));
`

XunNie, 2020-11-13:
Modules/planning/FastPlanner/plan_env/ThirdParty/sdf_tools/src/sdf_tools/sdf_builder.cpp 第80行左右, 修改参考:
``` c++
@@ -77,10 +77,12 @@ namespace sdf_tools
     {
         /* Builds a planning scene from XML string urdf and srdf descriptions */
         // Make the URDF model
-        boost::shared_ptr<urdf::Model> urdf_model(new urdf::Model());
+        //boost::shared_ptr<urdf::Model> urdf_model(new urdf::Model());
+        urdf::ModelSharedPtr urdf_model(new urdf::Model());
         urdf_model->initString(GenerateSDFComputeBotURDFString());
         // Make the SRDF model
-        boost::shared_ptr<srdf::Model> srdf_model(new srdf::Model());
+        //boost::shared_ptr<srdf::Model> srdf_model(new srdf::Model());
+        srdf::ModelSharedPtr srdf_model(new srdf::Model());
         srdf_model->initString(*urdf_model, GenerateSDFComputeBotSRDFString());
         // Make the planning scene
         planning_scene_ptr_.reset();
```

lzh-awesome, 2021-03-13:
Running command: "cmake /home/lzh/Prometheus/Modules/planning -DCATKIN_DEVEL_PREFIX=/home/lzh/Prometheus/devel -DCMAKE_INSTALL_PREFIX=/home/lzh/Prometheus/install -G Unix Makefiles" in "/home/lzh/Prometheus/build/planning"

CMake Error at CMakeLists.txt:1:
Parse error. Expected a command name, got unquoted argument with text
"/opt/ros/melodic/share/catkin/cmake/toplevel.cmake".

-- Configuring incomplete, errors occurred!
Invoking "cmake" failed‘

为什么这个模块cmakelist中只有一行？？？请问改怎么编译？

potato77, 2021-05-19:
你好，目前planning模块编译没问题，可以再次尝试编译，如果有问题，请重开issue提问。

closed 2021-05-19 by potato77