amov-lab/Prometheus #77: 运行 roslaunch prometheus_gazebo sitl.launch出错

YinghuiYou, 2020-11-08:
报错信息如下:
Resource not found: px4
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/ying/Prometheus/Modules/common/msgs
ROS path [2]=/home/ying/Prometheus/Modules/control
ROS path [3]=/home/ying/Prometheus/Modules/object_detection
ROS path [4]=/home/ying/Prometheus/Modules/mission
ROS path [5]=/home/ying/Prometheus/Modules/slam
ROS path [6]=/home/ying/Prometheus/Modules/planning
ROS path [7]=/home/ying/Prometheus/Simulator/gazebo_simulator
ROS path [8]=/home/ying/Prometheus/Experiment
ROS path [9]=/opt/ros/melodic/share
The traceback for the exception was written to the log file

Promethus的和Firmware_v110的编译都通过了

YinghuiYou, 2020-11-09:
问题已解决。
解决方法： 
1.使用echo $ROS_PACKAGE_PATH语句发现没有px4的路径没有被列入系统变量，
2. 将.zshrc 文件中的 export 语句放置在 source语句之后即可
3. 运行 source .zshrc

closed 2020-11-09 by YinghuiYou

bamboosdu, 2020-11-23:
Hi, I met error while I run  "roslaunch prometheus_gazebo sitl.launch"
Did you finally got the result of this command?

YinghuiYou, 2020-11-24:
> Hi, I met error while I run "roslaunch prometheus_gazebo sitl.launch"
> Did you finally got the result of this command?

I got the result. And I have viewed your issue, I never met such a problem during the time building the project.  I thought you may  not run ./compile_all.sh or other ./compile_*.sh before this command.  You can have a try,  but remember following the instruction on the wiki of this project. Best Wishes!