amov-lab/Prometheus #79: 关于运行fast_planner时出现的问题

BambiplayGit, 2020-11-17:
1.使用3dlidar导航建图时，飞机起点位置是world frame的（0，-10，0），但是建图是从world frame原点开始的。
[image]
这里原点产生的其实是world frame （0，-10，0）附近两根柱子。

2.使用rtab建图时
（1.）关于tf和frame报错：
Ignoring transform from authority "unknown_publisher" with frame_id and child_frame_id  "world" because they are the same
[image]
（2.）飞机初始位姿有问题，一开始就是朝world frame X轴正方向摆放，飞机只能横着飞，无法在gazebo内手动调整并且正常规划路径。
[image]

w407022008, 2021-01-26:
你可以试试把参考系map和world合并了，，odom里面的数据是相对于world的，

yi11122, 2021-03-31:
> 重度重图要求世界要求，奥多姆多姆·斯相关性世界，以及

请问可以稍微再详细一点么？对于这个坐标系的问题我也遇到了

w407022008, 2021-03-31:
@yi11122 试试改一下 [这一行](https://github.com/amov-lab/Prometheus/blob/c5888ba22b99d32ee9de2a4a4a644fbd4fac4d8d/Simulator/gazebo_simulator/launch_basic/tf_transform.launch#L18)

closed 2021-05-19 by potato77