amov-lab/Prometheus #81: "roslaunch prometheus_gazebo sitl.launch" error

bamboosdu, 2020-11-23:
While I run the command "roslaunch prometheus_gazebo sitl.launch"
I got error below.
```
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>PID Parameter <<<<<<<<<<<<<<<<<<<<<<<<<<<
Position Loop:  
Kp_xy : 0.8
Kp_z : 0.8
Velocity Loop:  
Kp_vxvy : 0.05
Kp_vz : 0.05
Ki_vxvy : 0.02
Ki_vz : 0.1
Kd_vxvy : 0.016
Kd_vz : 0
Limit:  
MPC_XY_VEL_MAX : 3
MPC_Z_VEL_MAX : 2
tilt_max : 20
MPC_THR_MIN : 0.1
MPC_THR_MAX : 0.9
Hover_throttle : 0.46
RLException: Roslaunch got a 'No such file or directory' error while attempting to run:

gnome-terminal --tab -- /src/Prometheus/devel/lib/prometheus_control/ground_station __name:=ground_station __log:=/root/.ros/log/88b551fc-2d77-11eb-8c5e-0242ac110002/ground_station-13.log

Please make sure that all the executables in this command exist and have
executable permission. This is often caused by a bad launch-prefix.
The traceback for the exception was written to the log file

```

bamboosdu, 2020-11-24:
I am sure that the "/src/Prometheus/devel/lib/prometheus_control/ground_station" exists.
What is the function of  **groud_station**?

473867143, 2020-11-26:
It does not refer to qgc, like screenshot 1 below:
[image]

Make sure you run the following code in the launch file，https://github.com/amov-lab/Prometheus/blob/master/Simulator/gazebo_simulator/launch/sitl.launch

	
	<node pkg="prometheus_control" type="ground_station" name="ground_station" output="screen" launch-prefix="gnome-terminal --tab --">	
	</node>

closed 2021-01-09 by potato77