amov-lab/Prometheus #83: A*算法路径规划的问题

Qing-erman, 2020-12-11:
在运行sitl_astar_3dlidar.launch时，无人机在垂直方向成正弦波方式运动，一直在震荡，请问这是正常的吗，是出了什么问题呢

potato77, 2021-01-09:
可以通过调参解决

closed 2021-01-09 by potato77

wcm-123, 2022-01-20:
> 在运行sitl_astar_3dlidar.launch时，无人机在垂直方向成正弦波方式运动，一直在震荡，请问这是正常的吗，是出了什么问题呢

你好，请教个问题
我将sitl_astar_3dlidar.launch文件中的障碍物膨胀距离设置成 0，如下

然后把global_planner.cpp文件中的安全距离设为0，如下
// 安全距离，若膨胀距离设置已考虑安全距离，建议此处设为0
nh.param("global_planner/safe_distance", safe_distance, 0);

再把地图的膨胀距离也设置成0，如下
// 地图膨胀距离，单位：米
nh.param("map/inflate", inflate_, 0);
改完之后进行的路径规划结果是，路径依然离障碍物很远，按道理会规划出一条比较靠近障碍物的路径，但是集中怎么规划都是与障碍物很远，有知道的小伙伴吗

wcm-123, 2022-01-20:
> 可以通过调参解决

你好，请教个问题
我将sitl_astar_3dlidar.launch文件中的障碍物膨胀距离设置成 0，如下

然后把global_planner.cpp文件中的安全距离设为0，如下
// 安全距离，若膨胀距离设置已考虑安全距离，建议此处设为0
nh.param("global_planner/safe_distance", safe_distance, 0);

再把地图的膨胀距离也设置成0，如下
// 地图膨胀距离，单位：米
nh.param("map/inflate", inflate_, 0);
改完之后进行的路径规划结果是，路径依然离障碍物很远，按道理会规划出一条比较靠近障碍物的路径，但是集中怎么规划都是与障碍物很远，有知道的小伙伴吗