amov-lab/Prometheus #96: /opt/ros/melodic/lib/libmavros.so：对‘mavconn::MAVConnUDP::get_remote_endpoint[abi:cxx11]() const’未定义的引用

958117216, 2021-01-21:
执行./compile_mission.sh 出现如下错误
/opt/ros/melodic/lib/libmavros.so：对‘mavconn::MAVConnUDP::get_remote_endpoint[abi:cxx11]() const’未定义的引用

[image]

473867143, 2021-01-22:
最近我也出现了这个问题，但是另外一台电脑是没有问题的。正在排除问题，初步怀疑该电脑环境可能存在问题。

958117216, 2021-01-22:
我的电脑是重装过eigen3的，安装mavros-extra出现过dpkg无法锁定eigen3，我不知道是不是这个引起的。
我分析他的报错，这个问题 好像是mission里的put_goal_from_qgc引起的，注释mission功能包的cmake文件的以下三行可解决这个报错

add_executable(pub_goal_from_qgc planning_mission/pub_goal_from_qgc.cpp)
add_dependencies(pub_goal_from_qgc prometheus_mission_gencpp)
target_link_libraries(pub_goal_from_qgc ${catkin_LIBRARIES})

但治标不治本。。。

473867143, 2021-01-22:
哈哈哈哈，这个是最近上传的代码。你可以先这样处理，我们再测试一下pub_goal_from_qgc.cpp问题，如果解决了，

958117216, 2021-01-22:
多谢多谢

473867143, 2021-03-30:
这个问题，我更新了一下系统就解决了。
`sudo apt-get update`
`sudo apt-get upgrade`

closed 2021-03-30 by 473867143