amov-lab/Prometheus #102: gazebo缺少模型的错误用什么办法解决呢

SIchenJi, 2021-01-28:
[Err] [InsertModelWidget.cc:405] Missing model.config for model "/home/sichenji/Prometheus/Simulator/gazebo_simulator/models/MultiUAV"
[Err] [InsertModelWidget.cc:405] Missing model.config for model "/home/sichenji/Prometheus/Simulator/gazebo_simulator/models/race_track"
[Err] [InsertModelWidget.cc:405] Missing model.config for model "/home/sichenji/Prometheus/Simulator/gazebo_simulator/models/texture"
[Err] [InsertModelWidget.cc:405] Missing model.config for model "/home/sichenji/Prometheus/Simulator/gazebo_simulator/models/texture_amov"
[Err] [InsertModelWidget.cc:405] Missing model.config for model "/home/sichenji/Prometheus/Simulator/gazebo_simulator/amov_models/MultiUAV"
INFO  [ecl/EKF] 3104000: EKF aligned, (baro hgt, IMU buf: 18, OBS buf: 14)
INFO  [ecl/EKF] 3104000: reset position to last known position
INFO  [ecl/EKF] 3104000: reset velocity to zero
[Err] [REST.cc:205] Error in REST request

libcurl: (51) SSL: no alternative certificate subject name matches target host name 'api.ignitionfuel.org'
INFO  [ecl/EKF] 5384000: reset position to GPS
INFO  [ecl/EKF] 5384000: reset velocity to GPS
INFO  [ecl/EKF] 5384000: starting GPS fusion
[Err] [ModelDatabase.cc:390] Unable to parse model.config for model[http://gazebosim.org/models/bin_4_dropping_task]

lzh-awesome, 2021-03-13:
路径文件吧 要把阿木的模型下载到本地

kavinwkp, 2021-03-24:
请问能说具体一点嘛 @lzh-awesome

potato77, 2021-05-19:
这个不影响，可忽略。

closed 2021-05-19 by potato77