amov-lab/Prometheus #103: Resource not found: px4

SIchenJi, 2021-01-28:
Resource not found: px4
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/sichenji/Prometheus/Modules/common/msgs
ROS path [2]=/home/sichenji/Prometheus/Modules/control
ROS path [3]=/home/sichenji/Prometheus/Modules/object_detection
ROS path [4]=/home/sichenji/Prometheus/Modules/mission
ROS path [5]=/home/sichenji/Prometheus/Modules/slam
ROS path [6]=/home/sichenji/Prometheus/Modules/planning
ROS path [7]=/home/sichenji/Prometheus/Modules/ground_station
ROS path [8]=/home/sichenji/Prometheus/Simulator/gazebo_simulator
ROS path [9]=/home/sichenji/Prometheus/Experiment
ROS path [10]=/opt/ros/melodic/share
ROS path [11]=/home/sichenji/promethues_px4
ROS path [12]=/home/sichenji/promethues_px4/Tools/sitl_gazebo
The traceback for the exception was written to the log file

我仔细查了一边路径，发现是拼写错误，
source /opt/ros/melodic/setup.bash
source /home/sichenji/Prometheus/devel/setup.bash
source /home/sichenji/prometheus_px4/Tools/setup_gazebo.bash /home/sichenji/promethues_px4 /home/sichenji/promethues_px4/build/amovlab_sitl_default
export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/home/sichenji/Prometheus/devel/lib
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/home/sichenji/Prometheus/Simulator/gazebo_simulator/models
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/home/sichenji/Prometheus/Simulator/gazebo_simulator/amov_models
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/home/sichenji/promethues_px4
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/home/sichenji/promethues_px4/Tools/sitl_gazebo
这里边所有的promethues都拼错了应该是prometheus

SIchenJi, 2021-01-28:
修改之后再
sichenji@sichenji:~$ roscd px4
就正常进入prometheus_px4
sichenji@sichenji:~/prometheus_px4$

473867143, 2021-01-29:
source的确很容易出现各种小问题，这里的顺序是有关系的，先source所有的Prometheus相关，后面source prometheus_px4相关，顺序错了，容易出现问题，你可以修改一下你的.bashrc为：

`source /opt/ros/melodic/setup.bash`
`source /home/sichenji/Prometheus/devel/setup.bash`
`export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:/home/sichenji/Prometheus/devel/lib`
`export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/home/sichenji/Prometheus/Simulator/gazebo_simulator/models`
`export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/home/sichenji/Prometheus/Simulator/gazebo_simulator/amov_models`
`source /home/sichenji/prometheus_px4/Tools/setup_gazebo.bash /home/sichenji/promethues_px4 /home/sichenji/promethues_px4/build/amovlab_sitl_default`
`export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/home/sichenji/promethues_px4`
`export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/home/sichenji/promethues_px4/Tools/sitl_gazebo`

closed 2021-02-09 by 473867143