amov-lab/Prometheus #105: eooro

b369147, 2021-02-01:
环境变量配置

    打开终端，并输入如下指令打开bashrc文件

    sudo gedit ~/.bashrc 

    在打开的文件中手动添加如下指令（以下若存在已添加过的命令，请勿重复添加），其中${your prometheus path}为Prometheus项目路径，${your px4 path}为安装PX4固件的路径。

    source ${your prometheus path}/Prometheus/devel/setup.bash
    export GAZEBO_PLUGIN_PATH=$GAZEBO_PLUGIN_PATH:${your prometheus path}/Prometheus/devel/lib
    export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:${your prometheus path}/Prometheus/Simulator/gazebo_simulator/models
    export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:${your prometheus path}/Prometheus/Simulator/gazebo_simulator/amov_models
    source ${your px4 path}/promethues_px4/Tools/setup_gazebo.bash ${your px4 path}/promethues_px4 ${your px4 path}/promethues_px4/build/amovlab_sitl_default
    export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:${your px4 path}/promethues_px4
    export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:${your px4 path}/promethues_px4/Tools/sitl_gazebo

备注：

    添加以上环境变量后，每次打开终端会出现配置好的路径，忽略即可。
    此步骤经常容易出错，请再三检查。（每一个路径都是有实际含义，请确保电脑中有该路径存在）

这一块为什么会错呢？
作者把这个写错了promethues_px4
应该是prometheus

473867143, 2021-02-01:
Prometheus是普罗米修斯代码 prometheus_px4是与之相对应的px4固件哦

b369147, 2021-02-01:
我晓得，但是你看一下那个px4的拼写已经错了

473867143, 2021-02-01:
哦哦哦！看到了，多谢

closed 2021-02-03 by 473867143