amov-lab/Prometheus #108: 运行失败

b369147, 2021-02-18:
请问：在运行roslaunch prometheus_gazebo sitl.launch的时候，出现
[gazebo-3] process has died [pid 28254, exit code 139, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver -e ode /home/a/Prometheus/Simulator/gazebo_simulator/worlds/empty.world __name:=gazebo __log:=/home/a/.ros/log/a010d7a4-71ce-11eb-bc90-50e0852e6daa/gazebo-3.log].
log file: /home/a/.ros/log/a010d7a4-71ce-11eb-bc90-50e0852e6daa/gazebo-3*.log
是怎么回事呢？

473867143, 2021-02-22:
可能是电脑性能不足，多运行几次。
也可能是有删改过，请恢复代码，重新编译，再次运行。

potato77, 2021-05-19:
在PX4固件中执行make amovlab_sitl_default gazebo试试有没有问题，如果没问题的话，那就重新运行launch文件几次（电脑差的话，gazebo经常启动失败）

closed 2021-05-19 by potato77

Chentao2000, 2021-12-11:
[image]
亲测有效！！