amov-lab/Prometheus #110: 关于运行出现 MODE: Unsupported FCU的问题

lzh-awesome, 2021-03-13:
运行launch文件出现以下错误，gazebo模型里用键盘控制飞机，飞机不起飞请问是什么原因呢？？

[ERROR] [1615635398.051488138, 13324.424000000]: MODE: Unsupported FCU
[ WARN] [1615635403.059604673, 13328.328000000]: CMD: Command 400 -- wait ack timeout

SiddharthJ25, 2021-03-29:
Hi, I am having the same issue. Did you resolve it? Also, my drone state in the second terminal is "Unconnected".

GanAH, 2021-04-01:
有大神出来解答吗，飞机无法起飞

lzh-awesome, 2021-04-21:
> 有大神出来解答吗，飞机无法起飞

我已经解决了可以看下这个链接：https://blog.csdn.net/zhihao8013/article/details/115334822?spm=1001.2014.3001.5501
或者这个链接：https://bbs.amovlab.com/forum.php?mod=viewthread&tid=27139&extra=

lzh-awesome, 2021-04-21:
@GanAHE @SiddharthJ25 我已经解决了可以看下这个链接：https://blog.csdn.net/zhihao8013/article/details/115334822?spm=1001.2014.3001.5501
或者这个链接：https://bbs.amovlab.com/forum.php?mod=viewthread&tid=27139&extra=

potato77, 2021-05-19:
请使用专用的prometheus_px4固件。
如果还有问题，建议重新编译一下固件。

closed 2021-05-19 by potato77

Xue-JW, 2022-02-10:
https://discuss.px4.io/t/pxh-shell-doesnt-launch-when-i-run-with-gazebo/19886/2