amov-lab/Prometheus #115: 路径规划中有关3dlidar不规划问题

Qing-erman, 2021-04-08:
planning模块，2d雷达的仿真均没问题，不会报错。
但是在3dlidar仿真中，局部点云输入：该话题定义在3Dlidar.sdf中 (PointCloud2) -->
        <remap from="cloud_in" to="/prometheus/sensors/3Dlidar_scan" /> 就不规划，
终端报错如下：
[octomap_server-16] process has died [pid 13035, exit code -11, cmd /opt/ros/melodic/lib/octomap_server/octomap_server_node cloud_in:=/prometheus/sensors/3Dlidar_scan __name:=octomap_server __log:=/home/amov/.ros/log/65b9aecc-981c-11eb-84c0-04d4c422f274/octomap_server-16.log].
log file: /home/amov/.ros/log/65b9aecc-981c-11eb-84c0-04d4c422f274/octomap_server-16*.log
终端显示“Need sensor info”，自我感觉是3dlidar传感器数据没有输入，或者3dlidar没有校准好，但是有关3dlidar的插件都装上了，不知道怎么解决运用3Dlida进行规划的问题。

lzh-awesome, 2021-04-21:
我也遇到了这个问题，请问你解决了吗？？一用到3dlidar就会报错

Qing-erman, 2021-04-22:
> 我也遇到了这个问题，请问你解决了吗？？一用到3dlidar就会报错
之前有阿木实验室的人说 好像是prometheus项目里的3dlidar有问题，就是有bug，但是我也不清楚怎么修改这个bug。

potato77, 2021-05-19:
请问问题还存在吗，运行的是哪个launch文件？

Qing-erman, 2021-05-19:
还存在的，运行的是sitl_astar_3dlidar.launch这个文件，prometheus版本是对应的p450的版本，请问要在哪里做修改呢

potato77, 2021-06-02:
你好，我这边运行sitl_astar_3dlidar.launch文件没有问题。是不是没有安装3d lidar的插件呢，或者octomap的插件

如：

sudo apt-get install ros-melodic-velodyne-gazebo-plugins

473867143, 2021-06-02:
安装一下octomap相关的ros插件

sudo apt-get install ros-melodic-octomap-*

closed 2021-06-05 by 473867143