amov-lab/Prometheus #118: 运行./compile_detection.sh出现的问题，不知道大家是否也存在，麻烦解答一下

Fllying, 2021-05-03:
wshe@wshe:~/px4_ws/src/Prometheus$ ./compile_detection.sh 
Base path: /home/wshe/px4_ws/src/Prometheus
Source space: /home/wshe/px4_ws/src/Prometheus/Modules/common/msgs
Build space: /home/wshe/px4_ws/src/Prometheus/build/msgs
Devel space: /home/wshe/px4_ws/src/Prometheus/devel
Install space: /home/wshe/px4_ws/src/Prometheus/install
####
#### Running command: "make cmake_check_build_system" in "/home/wshe/px4_ws/src/Prometheus/build/msgs"
####
####
#### Running command: "make -j8 -l8" in "/home/wshe/px4_ws/src/Prometheus/build/msgs"
####
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_SwarmCommand
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_IndoorSearch
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_Message
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_Formation
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_LogMessageDetection
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_CheckForObjectsAction
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_CheckForObjectsResult
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_ArucoInfo
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_PositionReference
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_ControlOutput
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_CheckForObjectsActionFeedback
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_LogMessagePlanning
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_LogMessage
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_ControlCommand
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_BoundingBoxes
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_CheckForObjectsGoal
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_BoundingBox
[  0%] Built target actionlib_msgs_generate_messages_lisp
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_CheckForObjectsActionResult
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_CheckForObjectsFeedback
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_LogMessageControl
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_AttitudeReference
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_CheckForObjectsActionGoal
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_MultiArucoInfo
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target actionlib_msgs_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_MultiDetectionInfo
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_DetectionInfo
[  0%] Built target _prometheus_msgs_generate_messages_check_deps_DroneState
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target actionlib_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target actionlib_msgs_generate_messages_nodejs
[ 19%] Built target prometheus_msgs_generate_messages_lisp
[ 39%] Built target prometheus_msgs_generate_messages_eus
[ 59%] Built target prometheus_msgs_generate_messages_cpp
[ 79%] Built target prometheus_msgs_generate_messages_py
[100%] Built target prometheus_msgs_generate_messages_nodejs
[100%] Built target prometheus_msgs_generate_messages
Base path: /home/wshe/px4_ws/src/Prometheus
Source space: /home/wshe/px4_ws/src/Prometheus/Modules/object_detection
Build space: /home/wshe/px4_ws/src/Prometheus/build/object_detection
Devel space: /home/wshe/px4_ws/src/Prometheus/devel
Install space: /home/wshe/px4_ws/src/Prometheus/install
####
[... 93 lines cut ...]
                                                 ^
In file included from /usr/local/include/opencv2/core/core_c.h:48:0,
                 from /usr/local/include/opencv2/core/utility.hpp:1234,
                 from /usr/local/include/opencv2/core.hpp:3286,
                 from /usr/local/include/opencv2/imgproc.hpp:46,
                 from /usr/local/include/opencv2/imgproc/imgproc.hpp:48,
                 from /home/wshe/px4_ws/src/Prometheus/Modules/object_detection/src/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:31,
                 from /home/wshe/px4_ws/src/Prometheus/Modules/object_detection/src/darknet_ros/src/YoloObjectDetector.cpp:10:
/usr/local/include/opencv2/core/types_c.h:327:1: note: candidate: _IplImage::_IplImage()
 _IplImage
 ^~~~~~~~~
/usr/local/include/opencv2/core/types_c.h:327:1: note:   candidate expects 0 arguments, 1 provided
/usr/local/include/opencv2/core/types_c.h:327:1: note: candidate: constexpr _IplImage::_IplImage(const _IplImage&)
/usr/local/include/opencv2/core/types_c.h:327:1: note:   no known conversion for argument 1 from ‘cv::Mat’ to ‘const _IplImage&’
/usr/local/include/opencv2/core/types_c.h:327:1: note: candidate: constexpr _IplImage::_IplImage(_IplImage&&)
/usr/local/include/opencv2/core/types_c.h:327:1: note:   no known conversion for argument 1 from ‘cv::Mat’ to ‘_IplImage&&’
CMakeFiles/darknet_ros_lib.dir/build.make:62: recipe for target 'CMakeFiles/darknet_ros_lib.dir/src/darknet_ros/src/YoloObjectDetector.cpp.o' failed
make[2]: *** [CMakeFiles/darknet_ros_lib.dir/src/darknet_ros/src/YoloObjectDetector.cpp.o] Error 1
CMakeFiles/Makefile2:1764: recipe for target 'CMakeFiles/darknet_ros_lib.dir/all' failed
make[1]: *** [CMakeFiles/darknet_ros_lib.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
[INFO] SUBMODULE object_detection_landing NOT EXIST, Skip it!
[INFO] SUBMODULE object_detection_yolov5openvino NOT EXIST, Skip it!

Fllying, 2021-05-03:
错误提示：
/home/wshe/px4_ws/src/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp: In function ‘void spire::tag_canny(cv::InputArray, cv::OutputArray, cv::OutputArray, cv::OutputArray, int, bool)’:
/home/wshe/px4_ws/src/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:1159:19: error: conversion from ‘cv::Mat’ to non-scalar type ‘CvMat’ requested
CvMat c_src = src, c_dst = edges.getMat();
^~~
/home/wshe/px4_ws/src/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:1159:45: error: conversion from ‘cv::Mat’ to non-scalar type ‘CvMat’ requested
CvMat c_src = src, c_dst = edges.getMat();
~~~~~~~~~~~~~^~
/home/wshe/px4_ws/src/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:1160:33: error: conversion from ‘cv::Mat’ to non-scalar type ‘CvMat’ requested
CvMat c_dx = sobel_x.getMat();
~~~~~~~~~~~~~~~^~
/home/wshe/px4_ws/src/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:1161:33: error: conversion from ‘cv::Mat’ to non-scalar type ‘CvMat’ requested
CvMat c_dy = sobel_y.getMat();

boyagesmile, 2021-05-07:
opencv版本问题，换成3.3.1版本就好了

Fllying, 2021-05-08:
@boyagesmile 谢谢，我换回了Ubuntu18自带的3.2.0就好了！我是在做其它项目的时候用了高版本的，所以编译这个项目的时候出错了！

potato77, 2021-05-19:
默认使用ubuntu1804自带的3.2.0是没问题的，如果是其他版本，请手动重装OPEN CV，详见https://github.com/amov-lab/Prometheus/wiki/Q&A%E6%B1%87%E6%80%BB

closed 2021-05-19 by potato77