amov-lab/Prometheus #128: 运行terminal_control无人机直接飞走

Jisooyaaa, 2021-06-24:
根据wiki上的测试命令：
roslaunch prometheus_gazebo sitl.launch
以及
rosrun prometheus_control terminal_control
终端未报错成功打开gazebo并且Ground station显示connected，通过terminal_control节点解锁并设置move模式，起飞后无人机直接飞走，无法飞向设定的航路点，在Ground station 终端中显示Target Info from PX4 的所有项都为nan。
测试其他模式也是如此，只要起飞无人机就直接飞走，随后显示out of the geo fence,the drone is landing......

473867143, 2021-06-25:
先999解锁并切换到offboard模式，然后再发1takeoff指令，你看这样子可以不？

Jisooyaaa, 2021-06-27:
> 先999解锁并切换到offboard模式，然后再发1takeoff指令，你看这样子可以不？

我重新编译产生的警告如下：
CMake Warning at /home/zdlab3/Prometheus/build/control/cmake/prometheus_control-genmsg.cmake:3 (message):
  Invoking generate_messages() without having added any message or service
  file before.

  You should either add add_message_files() and/or add_service_files() calls
  or remove the invocation of generate_messages().
---------------------------------------------------------------------------------------------------
CMake Warning (dev) at CMakeLists.txt:311 (add_dependencies):
  Policy CMP0046 is not set: Error on non-existent dependency in
  add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.
------------------------------------------------------------------------------------------------------
fastplanning模块编译也报错：
/home/zdlab3/Prometheus/Modules/planning/FastPlanner/plan_env/ThirdParty/sdf_tools/src/sdf_tools/sdf_builder.cpp:87:134: error: no matching function for call to ‘planning_scene::PlanningScene::PlanningScene(boost::shared_ptr<urdf::Model>&, boost::shared_ptr<srdf::Model>&)’
         planning_scene_ptr_ = std::shared_ptr<planning_scene::PlanningScene>(new planning_scene::PlanningScene(urdf_model, srdf_model));

Jisooyaaa, 2021-06-27:
> 先999解锁并切换到offboard模式，然后再发1takeoff指令，你看这样子可以不？

我觉得应该是这里的问题：
CMake Warning at /home/zdlab3/Prometheus/build/control/cmake/prometheus_control-genmsg.cmake:3 (message):
  Invoking generate_messages() without having added any message or service
  file before.

  You should either add add_message_files() and/or add_service_files() calls
  or remove the invocation of generate_messages().
Call Stack (most recent call first):
  /opt/ros/melodic/share/genmsg/cmake/genmsg-extras.cmake:307 (include)
  CMakeLists.txt:36 (generate_messages)

-- prometheus_control: 0 messages, 0 services
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zdlab3/Prometheus/build/control
为什么0 message呢？

potato77, 2021-06-30:
> 根据wiki上的测试命令：
> roslaunch prometheus_gazebo sitl.launch
> 以及
> rosrun prometheus_control terminal_control
> 终端未报错成功打开gazebo并且Ground [...]

这种情况是，没有将定位信息发送至PX4，导致其位置估算不准，就直接飞上天了。
请问使用了prometheus_px4固件了吗？
如果使用了，请先echo 看一下/mavros/vision_estimation/pose这个信息是否和地面站打印的位置信息吻合。
或者可以发一下节点图或者地面站截图

closed 2021-08-11 by potato77