amov-lab/Prometheus #132: ROS多机问题

lzh-awesome, 2021-07-14:
我想尝试100架的仿真，我先用ROS多机尝试了5架的仿真
我目前已经用ROS多机将三台电脑连在同一局域网下，已经配置好了主机IP，从机IP。
我现在ROS_MASTER主机上启动了roscore 以及roslaunch sitl_formation_5uav.launch，
在从机端分别启动了规划、控制、队形变换的节点，在主机端的gazebo中已经实现了队形的变换。
但是我有一个疑问？
主机上启动的飞机初始化的launch文件启动了PX4_sitl模块和simulator模块这两个必备模块，这个PX4_sitl模块能否也放到从机端运行呢，还是只能绑定在一起？PX4_sitl模块很占电脑资源，飞机模型是否也会影响电脑资源呢，我想尝试最基本的iris模型
我想把飞机规模扩大成100架，，请问我的理解正确吗？可以给我一点建议吗？

potato77, 2021-07-14:
PX4_sitl模块可以在从机中启动，因为gazebo界面就是起到一个显示功能。

但是，100架似乎太多了，问题很多。

1、你如果一定要用PX4 sitl的话，那你只能去尝试一下了。我这边试过单个电脑最多20台飞机左右（建议分批启动，不然会卡），即使你用多个电脑，你的gazebo显示界面应该是只能有一个，启动很多架之后，gazebo的realtime_factor就会变得很小，整个仿真就乱套了。我最后放弃了这种方式。

2、给你一个跳过PX4 sitl的思路。自己写一个模拟无人机运动的脚本（订阅控制指令，输出无人机状态），输出的状态用gazebo/set_model_state更新无人机的位置。（我说的很简单啊，但是我当时也开发了很久）

两个方法坑都很多，建议还是别跑100架了，10架差不多了

lzh-awesome, 2021-07-20:
感谢戚博的回复，
我目前想通过ROS多机分批启动gazebo，即在ROSmaster主机中只启动gazebo_sim界面，在从机中启动PX4_sitl模块。
我目前尝试了20架的仿真，修改的初始化的launch文件：
1）主机launch文件（启动了gazebo_sim和10架UAV的px4_sitl）
2）从机的launch文件(我删除了gazebo_sim的配置),px4_sitl模块可以启动，但是在主机gazebo上并不能加载剩余的11-20号飞机？
请问如何分批启动gazebo的无人机呢，分批启动无人机是否支持ROS多机？？是不是我的gazebo配置有问题，还是我哪里理解错误？

potato77, 2021-07-21:
gazebo是支持ros多机的。
检查一下在从机的终端输出，px4_sitl是否启动成功，以及另外在主机中echo一下能否收到第11架飞机的mavros消息等等，确保你的ros多机是设置正确的。

lzh-awesome, 2021-08-02:
感谢戚博的回复，
我目前在主机端gazebo分批启动uav已经成功，并且uav能够连上mavros，即在主机端依次启动gazebo界面，10架uav，另外10架uav（主机端总共20架已经成功）。
我想继续把规模做大，但是我在尝试：
1）主机gazebo+10架uav的px4_sitl
2)  从机10架uav的px4_sitl时，
发现主机端1~10架的px4_sitl启动成功，并且成功连上mavros，在gazebo界面中成功显示；
从机端的11~20架的px4_sitl并没有完全启动成功，卡在MAVROS started这里，gazebo中只增加了1架uav，在主机中echo /mavros/states 时，发现1~10架都能连上，11~20都无法连接上；

从机端px4_sitl终端输出如下：
====================================================
开始时：
ROS_MASTER_URI=http://192.168.3.112:11311
process[uav11/sitl_11-1]: started with pid [73943]
INFO  [px4] Creating symlink /home/lzh/dorca_ws/src/d-orca/Firmware/ROMFS/px4fmu_common -> /home/lzh/.ros/etc
0 WARNING: setRealtimeSched failed (not run as root?)

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 11
process[uav11/iris_11_spawn-2]: started with pid [73950]
process[uav11/mavros-3]: started with pid [73961]
+ MAV_SYS_ID: curr: 5 -> new: 12
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4571
process[uav12/sitl_12-4]: started with pid [73989]
INFO  [px4] Creating symlink /home/lzh/dorca_ws/src/d-orca/Firmware/ROMFS/px4fmu_common -> /home/lzh/.ros/etc
0 WARNING: setRealtimeSched failed (not run as root?)

==================================================================
结束时（卡在这里不继续运行了）：
[ INFO] [1627868505.525849422]: Known MAVLink dialects: common ardupilotmega ASLUAV all autoquad icarous matrixpilot paparazzi standard uAvionix ualberta
[ INFO] [1627868505.525860076]: MAVROS started. MY ID 1.240, TARGET ID 18.1

我目前不能把PX4_sitl模块在从机中启动，请问是我哪里理解错了吗？