amov-lab/Prometheus #139: fast-planner仿真，在指定终点坐标时报错  When running Fast-Planner, the program crashed while specifying the destination location

brain726, 2021-08-02:
fast-planner仿真，在指定终点坐标时报错，我最开始指定终点使用的是rviz的插件3D Nav Goal，使用后程序报错，后来我修改launch文件，改成使用终端指定终点，程序报错，两次报错相同。错误如下：
[image]
[image]
When running Fast-Planner, the program crashed while specifying the destination location。First time, I used '3D Nav Goal' in rviz but the program crashed. After that, I used terminal to input the location, but the program output the same error.

pzx-Id, 2021-08-06:
@potato77 你好，ego-planner的launch运行文件是哪个呢？

pzx-Id, 2021-08-06:
@potato77 
[image]
请问这个是什么问题呢？ 该怎么解决阿。

brain726, 2021-08-06:
您好，感谢您的回复，我运行的是sitl_fast_planning_3dlidar.launch

brain726, 2021-08-06:
您好，不好意思，刚才看错了，那个是fastplanner的launch文件
我这里ego-planner也不能运行，但是重新从github上下载了ego-planner源码可以运行rviz下的仿真，结合p450上的实现勉强可以学习

pzx-Id, 2021-08-06:
您好，您运行是报什么错呢？
我刚运行了，有下面的错误：
[image]
然后我想知道您这个ego-planner,有结合p450在gazebo上仿真吗？
想跟你学习学习。

brain726, 2021-08-06:
这个是因为有的功能包没加入bashrc，我也经常遇到这个问题，你是自己配的prometheus吗，还是买的阿木配好的机器

brain726, 2021-08-06:
p450跑不了仿真，arm的cpu太差了，建议i7或以上的机器

pzx-Id, 2021-08-06:
自己配的，目前还没买实物，想先仿真跑一下先

brain726, 2021-08-06:
应该是bashrc没配置对，建议参考这篇文章
(14条消息) 阿木实验室的普罗米修斯仿真平台搭建的难点和坑点_Sichen Ji的博客-CSDN博客

pzx-Id, 2021-08-07:
好的，谢谢。其他好像都没问题，就是跑ego-planner的时候出现了。