amov-lab/Prometheus #140: 从mavros中读取飞机姿态问题

RBKing20, 2021-08-02:
px4_pos_estimator中发布的DroneState的attitude_q是从state_from_mavros中的att_cb中获取到/mavros/imu/data，为什么不从/mavros/local_position/pose的orientation获取姿态呢，感觉后者在融完后会更稳定和准确一些鸭

w407022008, 2021-08-02:
这两个应该是同一个数据。[/mavros/local_position/pose/orientation](http://docs.ros.org/en/jade/api/mavros/html/local__position_8cpp_source.html#:~:text=auto%20enu_orientation_msg%20%3D%20uas-%3Eget_attitude_orientation()%3B) 指向[/mavros/imu/data](http://docs.ros.org/en/jade/api/mavros/html/uas__data_8cpp_source.html#:~:text=return%20imu_data-%3Eorientation%3B)
另外，在px4中，vehicle_local_position.msg消息本身是不带有姿态信息的。在ekf2.cpp文件中，也是首先发布vehicle_attitude.msg消息随后发布vehicle_local_position.msg。

RBKing20, 2021-08-02:
> 这两个应该是同一个数据。[/mavros/local_position/pose/orientation](http://docs.ros.org/en/jade/api/mavros/html/local__position_8cpp_source.html#:~:text=auto%20enu_orientation_msg%20%3D%20uas-%3Eget_attitude_orient [...]

二者应该还是有区别，我试着在仿真下打印出数据，两者比较接近但并非完全一样，如果使用了vicon或t265，那应该就不是一样的吧？

w407022008, 2021-08-02:
我查看了一下px4源码，姿态数据仅由imu提供的，其他外部传感器包括vicon t265作为外部视觉仅提供heading修正。因此，在增加外部传感器的情况下，heading数据会比仅使用imu中的mega更加精确。不过这并不影响MAVLink数据， 因为Prometheus接受的姿态信息是来自MAVLink的Attitude(#30)，这个消息发布自px4系统中的vehicle_attitude消息，而这个消息本身就是修正后的消息。另外，LOCAL_POSITION_NED ( #32 )消息本身不包含姿态，/mavros/local_position/pose/orientation这个消息也是来自Attitude(#30)的。

RBKing20, 2021-08-02:
@w407022008 明白了，非常感谢

closed 2021-08-02 by RBKing20