amov-lab/Prometheus #142: 启动prometheus_gazebo sitl.launch仿真时mavlink错误

MaverickCCC, 2021-08-03:
[image]
**图1**

[image]
**图2**

Gazebo仿真运行测试，启动launch文件后提示图1中udp_port的错误，终端控制terminal_control 输入指令后无法解锁，且提示图2中FCU错误。
想请教一下可能是哪里出问题了？

（自己重新安装编译过好几次了，还是有问题）

potato77, 2021-08-04:
这个报错我也遇到过，但似乎不影响飞行。

具体是启动的那个launch文件，不能打印什么信息呢？ @Boshen97

MaverickCCC, 2021-08-05:
感谢戚博的回复，udp这个问题不大我就放心了。

我具体启动的是swarm_4uav那个launch文件，我看到你集群直播课里启动后有3个显示终端和1个控制终端，课上你说第三个显示终端会打印出无人机的一些信息。

我这边虽然可以飞行，但唯独少了第三个显示终端，除了这个udp端口的报错外，没有其他报错，所以不知道什么原因导致这个终端没有出现，不知道戚博是否遇到过？

| |
陈兆通
|
|
***@***.***
|
签名由网易邮箱大师定制

potato77, 2021-08-08:
你看看终端是否打印齐了四架飞机的数据呢？
如果功能上没有问题的话，可能是后来精简了一下终端打印吧。
@Boshen97 协助解决一下吧

closed 2021-08-09 by MaverickCCC