amov-lab/Prometheus #144: 运行ego_planner出问题。

pzx-Id, 2021-08-07:
[image]
[image]
运行ego_planner发现上面的错误，控制不了无人机。（可能是mavros没连上）
不知道是我的运行问题还是怎么的。
该怎么解决呢？

potato77, 2021-08-08:
就是mavros没连接上，切换px4代码分支了么？

pzx-Id, 2021-08-08:
应该是没有切换px4代码分支。应该怎么切换px4代码分支呢？

pzx-Id, 2021-08-08:
切换了px4代码分支后还是存在这个问题。 @potato77

potato77, 2021-08-11:
我这边仿真是正常的，你再检查一下PX4脚本还有启动文件吧

[image]

pzx-Id, 2021-08-12:
好的，谢谢。我运行的时候，第一次是mavros正常的，但是不能控制无人机。
再第二次运行后就出现了之前那种情况。

pzx-Id, 2021-08-14:
再次打扰您一下，我这个连接都没问题，但是无人机没有反应，飞不起来。 状态是connected，但是是disarmed，这个应该怎么解决呢？

closed 2021-08-21 by pzx-Id