amov-lab/Prometheus #147: 编译planning、slam等模块时的报错问题（已看过所有相关issues）

MaverickCCC, 2021-08-12:
planning和ego编译错误提示在：[ 97%] Built target traj_server  为图1和2所示。

slam编译错误提示：./compile_slam.sh: 第 5 行: cd: Modules/slam/config/Vocabulary: 没有那个文件或目录
tar: ORBvoc.txt.tar.gz：无法 open: 没有那个文件或目录
tar: Error is not recoverable: exiting now

[image]
**图1**
[image]
**图2**

 
我已查阅过其他功能包编译的issues，同时试过其余模块单独编译没有其他问题。
想知道以上问题可以如何解决，谢谢！

tugepaopaoo, 2021-08-17:
遇到了相同的问题

tugepaopaoo, 2021-08-18:
我这边在编译planning和ego模块出现“未定义的引用”报错后，将Prometheus/build中编译生成的ego_planner、planning文件夹分别删除，然后再重新编译，可以编译成功。

473867143, 2021-08-18:
> planning和ego编译错误提示在：[ 97%] Built target traj_server 为图1和2所示。
>
> slam编译错误提示：./compile_slam.sh: 第 5 行: cd: Modules/slam/config/Vocabulary: 没有那个文件或目录
> tar: ORBvoc.txt.tar.gz：无法 open: 没有那个文件或目录
> tar: Error [...]

slam编译报错不影响使用。

473867143, 2021-08-18:
> 我这边在编译planning和ego模块出现“未定义的引用”报错后，将Prometheus/build中编译生成的ego_planner、planning文件夹分别删除，然后再重新编译，可以编译成功。

对，最好把build和devel编译文件删掉，然后再重新编译，先编译planning再编译ego

closed 2021-10-20 by potato77