amov-lab/Prometheus #159: 使用 RTSP 推流ROS 图像失败

wuhaixiang, 2021-09-28:
环境： vmware
系统：ubuntu 18.04 
运行脚本 rtsp_steam.launch
错误地址：
def main(width,height,port,factory_name):
    global out_send
    out_send = cv2.VideoWriter('appsrc is-live=true ! videoconvert ! \
                                omxh264enc bitrate=12000000 ! video/x-h264, \
                                stream-format=byte-stream ! rtph264pay pt=96 ! \
                                udpsink host=192.168.86.141 port=5400 async=false',
                                cv2.CAP_GSTREAMER,30, (width,height), True)
 
    if not out_send.isOpened():
        print('VideoWriter not opened')
        exit(0)
提示 out_send 没有打开，程序退出

huangwen0907, 2021-09-28:
I have the same problem, if you have some good idea, please let me know. TKS.

likehengqq, 2021-09-28:
仅可以用于Nvidia TX2与NX平台，通过Nvidia Gstreamer进行编码

potato77, 2021-09-29:
X86可以试一下这个教程，https://github.com/amov-lab/Prometheus/wiki/%E5%B0%86%E6%A3%80%E6%B5%8B%E7%BB%93%E6%9E%9C(ROS%E5%9B%BE%E5%83%8F)%E6%8E%A8%E6%B5%81%E5%88%B0%E5%9C%B0%E9%9D%A2%E7%AB%99

closed 2021-10-18 by 473867143