amov-lab/Prometheus #173: 跟着教程一步一步走，走到make amovlab_sitl_default gazebo的时候报了如下的错

J1dan, 2021-10-16:
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/jidan/.gazebo/models/gazebo_models"
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/jidan/Prometheus/Simulator/gazebo_simulator/models/MultiUAV"
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/jidan/Prometheus/Simulator/gazebo_simulator/models/race_track"
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/jidan/Prometheus/Simulator/gazebo_simulator/models/texture"
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/jidan/Prometheus/Simulator/gazebo_simulator/models/texture_amov"
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/jidan/Prometheus/Simulator/gazebo_simulator/amov_models/MultiUAV"
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/jidan/Prometheus/Simulator/gazebo_simulator/amov_models/Multisolo"
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/jidan/Prometheus/Simulator/gazebo_simulator/amov_models/texture_aruco"
[Err] [InsertModelWidget.cc:426] Missing model.config for model "/home/jidan/Prometheus/Simulator/gazebo_simulator/amov_models/texture_prometheus"

我好奇的是本来这个路径下就没有model.config这个文件啊，为什么还报错?在编译Prometheuse文件夹的时候没有问题啊？

虽然报了错，但是gazebo还是成功启动了，并且忽略掉的话，接着跟教程走，最后一步roslaunch prometheus_gazebo sitl.launch也没有任何问题，但是在运行 roslaunch prometheus_gazebo sitl_color_line_following.launch这样其他的程序的时候会报如下的错：
[ERROR] [1634396629.582456, 0.000000]: Invalid XML: XML or text declaration not at start of entity: line 2, column 0
[vehicle_spawn_jidan_33400_2203756662665692012-5] process has died [pid 33473, exit code 1, cmd /opt/ros/melodic/lib/gazebo_ros/spawn_model -sdf -file /home/jidan/Prometheus/Simulator/gazebo_simulator/amov_models/p450_monocular/p450_monocular.sdf -model p450_monocular -x 0.0 -y 0.0 -z 0.06 -R 0.0 -P 0.0 -Y 0.0 __name:=vehicle_spawn_jidan_33400_2203756662665692012 __log:=/home/jidan/.ros/log/43a1cdf8-2e92-11ec-9af7-000c297d07ac/vehicle_spawn_jidan_33400_2203756662665692012-5.log].
log file: /home/jidan/.ros/log/43a1cdf8-2e92-11ec-9af7-000c297d07ac/vehicle_spawn_jidan_33400_2203756662665692012-5*.log
并且加载不出无人机模型（其他模型正常）。
麻烦解答一下我的问题，可能过两天就要买你们的无人机了

J1dan, 2021-10-17:
我自己解决了。。。问题出在报错的xml文件，找到之后删掉第一行，现在能正常运行其他程序了

473867143, 2021-10-20:
在Prometheus/Simulator/gazebo_simulator/amov_models/p450_monocular/p450_monocular.sdf中将第一行中文注释注释掉，由于中文导致的模型加载问题，已更新

chengfeng911, 2021-12-31:
> 我自己解决了。。。问题出在报错的xml文件，找到之后删掉第一行，现在能正常运行其他程序了

你好，我和你出现了一模一样的问题，请问这个xml文件在哪阿，我一直没有找到

J1dan, 2022-01-01:
> 

/home/jidan/Prometheus/Simulator/gazebo_simulator/amov_models/p450_monocular/p450_monocular.sdf应该是这个，好久前遇到的问题 我也记不太清了

wcm-123, 2022-01-16:
> 
我的p450_monocular.sdf文件里面前面没有中文注释,但依然报这种相同的错误，不解

`<?xml version="1.0"?>
<sdf version='1.6'>

  <model name='p450_monocular'>
  
    <include>
      <uri>model://p450</uri> 
    </include>

   
    <link name='Monocular_down_link'>
      <pose>0 0 -0.1 0 1.5707963 0</pose> 
      
      <inertial>
        <mass>0.001</mass> 
         
        <inertia>
          <ixx>4.15e-6</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>2.407e-6</iyy>
          <iyz>0</iyz>
          <izz>2.407e-6</izz>
        </inertia>
      </inertial>
      
      <visual name='visual'>
      
......`

wcm-123, 2022-01-16:
> > 
> 
> /home/jidan/Prometheus/Simulator/gazebo_simulator/amov_models/p450_monocular/p450_monocular.sdf应该是这个，好久前遇到的问题 我也记不太清了

我的p450_monocular.sdf文件里面前面没有中文注释，如下,但依然报这种相同的错误，
`<?xml version="1.0"?>
<sdf version='1.6'>

  <model name='p450_monocular'>
  
    <include>
      <uri>model://p450</uri> 
    </include>

   
    <link name='Monocular_down_link'>
      <pose>0 0 -0.1 0 1.5707963 0</pose> 
      `

wcm-123, 2022-01-16:
> > 
> 
> /home/jidan/Prometheus/Simulator/gazebo_simulator/amov_models/p450_monocular/p450_monocular.sdf应该是这个，好久前遇到的问题 我也记不太清了
我的p450_monocular.sdf文件里面前面没有中文注释，如下,但依然报这种相同的错误，
`<?xml version="1.0"?>
<sdf version='1.6'>
`

wcm-123, 2022-01-16:
> 

我的/p450_monocular.sdf前面两行是这样的
<?xml version="1.0"?>
<sdf version='1.6'>
没有中文注释，但依然报这种相同的错误

pythoncjava9, 2022-01-17:
我也报了相同的错误，但/p450_monocular.sdf前面两行是这样的：
<?xml version="1.0"?>
<sdf version='1.6'>
该怎么解决呢？

123any, 2022-03-13:
> 我也报了相同的错误，但/p450_monocular.sdf前面两行是这样的： `<?xml version="1.0"?> <sdf version='1.6'>` 该怎么解决呢？

请问一下有解决这个问题吗，我也是这个报错

m1284, 2022-07-02:
请问一下有解决这个问题吗，我也是这个报错，这个sdf文件开头也没有中文注释

liu3238, 2022-07-07:
解决了吗，我现在也是这样