amov-lab/Prometheus #188: ./compile_detection.sh   error

imposin, 2021-11-17:
CMake Error at /opt/ros/melodic/share/cv_bridge/cmake/cv_bridgeConfig.cmake:113 (message):
  Project 'cv_bridge' specifies '/usr/include/opencv' as an include dir,
  which is not found.  It does neither exist as an absolute directory nor in
  '${{prefix}}//usr/include/opencv'.  Check the issue tracker
  'https://github.com/ros-perception/vision_opencv/issues' and consider
  creating a ticket if the problem has not been reported yet.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  CMakeLists.txt:51 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/nano/Prometheus/build/object_detection/CMakeFiles/CMakeOutput.log".
See also "/home/nano/Prometheus/build/object_detection/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

imposin, 2021-11-17:
new issue  jetson nano 

/home/nano/Prometheus/Modules/object_detection/cpp_nodes/calibrate_camera_charuco.cpp:10:10: fatal error: opencv2/aruco/charuco.hpp: 没有那个文件或目录
 #include <opencv2/aruco/charuco.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/calibrate_camera_charuco.dir/build.make:62: recipe for target 'CMakeFiles/calibrate_camera_charuco.dir/cpp_nodes/calibrate_camera_charuco.cpp.o' failed
make[2]: *** [CMakeFiles/calibrate_camera_charuco.dir/cpp_nodes/calibrate_camera_charuco.cpp.o] Error 1
CMakeFiles/Makefile2:141: recipe for target 'CMakeFiles/calibrate_camera_charuco.dir/all' failed
make[1]: *** [CMakeFiles/calibrate_camera_charuco.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
/home/nano/Prometheus/Modules/object_detection/cpp_nodes/aruco_det_v2.cpp:41:10: fatal error: opencv2/aruco.hpp: 没有那个文件或目录
 #include <opencv2/aruco.hpp>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/aruco_det_v2.dir/build.make:62: recipe for target 'CMakeFiles/aruco_det_v2.dir/cpp_nodes/aruco_det_v2.cpp.o' failed
make[2]: *** [CMakeFiles/aruco_det_v2.dir/cpp_nodes/aruco_det_v2.cpp.o] Error 1
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/aruco_det_v2.dir/all' failed
make[1]: *** [CMakeFiles/aruco_det_v2.dir/all] Error 2
/home/nano/Prometheus/Modules/object_detection/cpp_nodes/landpad_det.cpp:41:10: fatal error: opencv2/aruco.hpp: 没有那个文件或目录
 #include <opencv2/aruco.hpp>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/landpad_det.dir/build.make:62: recipe for target 'CMakeFiles/landpad_det.dir/cpp_nodes/landpad_det.cpp.o' failed
make[2]: *** [CMakeFiles/landpad_det.dir/cpp_nodes/landpad_det.cpp.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/landpad_det.dir/all' failed
make[1]: *** [CMakeFiles/landpad_det.dir/all] Error 2
/home/nano/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp: In function ‘void spire::cv_canny(const void*, void*, void*, void*, int, bool, double)’:
/home/nano/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:630:5: error: ‘CvMat’ was not declared in this scope
     CvMat srcstub, *src = cvGetMat(srcarr, &srcstub); // IplImage to cvMat
     ^~~~~
/home/nano/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:631:11: error: expected ‘;’ before ‘dststub’
     CvMat dststub, *dst = cvGetMat(dstarr, &dststub);
           ^~~~~~~
/home/nano/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:633:11: error: expected ‘;’ before ‘dxstub’
     CvMat dxstub, *dx = cvGetMat(dxarr, &dxstub);
           ^~~~~~
/home/nano/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:634:11: error: expected ‘;’ before ‘dystub’
     CvMat dystub, *dy = cvGetMat(dyarr, &dystub);
           ^~~~~~
In file included from /usr/include/opencv4/opencv2/core.hpp:52:0,
                 from /usr/include/opencv4/opencv2/opencv.hpp:52,
                 from /home/nano/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.h:8,
                 from /home/nano/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:1:
/home/nano/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:636:21: error: ‘src’ was not declared in this scope
     if (CV_MAT_TYPE(src->type) != CV_8UC1 ||
                     ^
/home/nano/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:636:21: note: suggested alternative: ‘sqrt’
/home/nano/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:637:21: error: ‘dst’ was not declared in this scope
         CV_MAT_TYPE(dst->type) != CV_8UC1 ||
                     ^
/home/nano/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:638:21: error: ‘dx’ was not declared in this scope
         CV_MAT_TYPE(dx->type) != CV_16SC1 ||
                     ^
/home/nano/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:639:21: error: ‘dy’ was not declared in this scope
         CV_MAT_TYPE(dy->type) != CV_16SC1)
                     ^
In file included from /usr/include/opencv4/opencv2/core.hpp:54:0,
                 from /usr/include/opencv4/opencv2/opencv.hpp:52,
                 from /home/nano/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.h:8,
                 from /home/nano/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:1:
/home/nano/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:640:18: error: ‘CV_StsUnsupportedFormat’ was not declared in this scope
         CV_Error(CV_StsUnsupportedFormat, "");
                  ^
/home/nano/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:642:26: error: ‘src’ was not declared in this scope
     if (!CV_ARE_SIZES_EQ(src, dst))
                          ^~~
/home/nano/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:642:26: note: suggested alternative: ‘sqrt’
     if (!CV_ARE_SIZES_EQ(src, dst))
                          ^~~
[... 136 lines cut ...]
/home/nano/Prometheus/Modules/object_detection/src/ellipse_lib/spire_ellipsedetector.cpp:4350:47: warning: narrowing conversion of ‘(p3.cv::Point_<int>::x * p3.cv::Point_<int>::y)’ from ‘int’ to ‘float’ inside { } [-Wnarrowing]
                     p3.x*p3.x, p3.y*p3.y, p3.x*p3.y, p3.x, p3.y,
[... cut ...]
                     2 * p3.x,    2 * t3*p3.y, t3*p3.x + p3.y, 1,   t3 };
                     ~~^~~~~~
CMakeFiles/ellipselib.dir/build.make:62: recipe for target 'CMakeFiles/ellipselib.dir/src/ellipse_lib/spire_ellipsedetector.cpp.o' failed
make[2]: *** [CMakeFiles/ellipselib.dir/src/ellipse_lib/spire_ellipsedetector.cpp.o] Error 1
CMakeFiles/Makefile2:252: recipe for target 'CMakeFiles/ellipselib.dir/all' failed
make[1]: *** [CMakeFiles/ellipselib.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

potato77, 2021-11-17:
OPENCV没有源码编译，推荐版本是3.3.1
参考：https://github.com/amov-lab/Prometheus/wiki/Q&A%E6%B1%87%E6%80%BB

imposin, 2021-11-17:
戚博  我还想问  实际飞行时  位置估计模块  没有vison  没有激光slam  没有t265   什么都没有 只有gps   这个模块的参数选择什么

	<node pkg="prometheus_control" type="px4_pos_estimator" name="px4_pos_estimator" output="screen">
		
		<param name="input_source" value="0" />
		
		<param name="object_name" value="UAV"/>
		<param name="rate_hz" value="20.0"/>
	</node>

imposin, 2021-11-18:
我好像找到了 

  ros::init(argc, argv, "px4_pos_estimator");
    ros::NodeHandle nh("~");

    //读取参数表中的参数
    // 定位数据输入源 0 for vicon， 1 for 激光SLAM, 2 for gazebo ground truth, 3 for T265 ,  9 for outdoor 
    nh.param<int>("input_source", input_source, 0);
    // 动作捕捉设备中设定的刚体名字
    nh.param<string>("object_name", object_name, "UAV");
    //　程序执行频率
    nh.param<float>("rate_hz", rate_hz, 20);
    //　定位设备偏移量
    nh.param<float>("offset_x", pos_offset[0], 0);
    nh.param<float>("offset_y", pos_offset[1], 0);
    nh.param<float>("offset_z", pos_offset[2], 0);
    nh.param<float>("offset_yaw", yaw_offset, 0);

选9就行了吧

closed 2021-11-24 by 473867143