amov-lab/Prometheus #192: 执行make amovlab_sitl_default gazebo报错

wcm-123, 2021-11-24:
Failed to import numpy: No module named 'numpy'

You may need to install it using:
    pip3 install --user numpy

src/lib/mixer/MultirotorMixer/CMakeFiles/mixer_gen.dir/build.make:83: recipe for target 'src/lib/mixer/MultirotorMixer/mixer_multirotor.generated.h' failed
make[4]: *** [src/lib/mixer/MultirotorMixer/mixer_multirotor.generated.h] Error 1
make[4]: *** 正在等待未完成的任务....
[  2%] Linking CXX static library libdrivers_board.a
Failed to import numpy: No module named 'numpy'

You may need to install it using:
    pip3 install --user numpy

src/lib/mixer/MultirotorMixer/CMakeFiles/mixer_gen.dir/build.make:109: recipe for target 'src/lib/mixer/MultirotorMixer/mixer_multirotor_normalized.generated.h' failed
make[4]: *** [src/lib/mixer/MultirotorMixer/mixer_multirotor_normalized.generated.h] Error 1
CMakeFiles/Makefile2:3769: recipe for target 'src/lib/mixer/MultirotorMixer/CMakeFiles/mixer_gen.dir/all' failed
make[3]: *** [src/lib/mixer/MultirotorMixer/CMakeFiles/mixer_gen.dir/all] Error 2
make[3]: *** 正在等待未完成的任务....
Scanning dependencies of target mixer_gen_6dof
[  3%] Generating mixer_multirotor_6dof.generated.h
[  3%] Built target drivers_board
Failed to import numpy: No module named 'numpy'

You may need to install it using:
    pip3 install --user numpy

src/lib/mixer/MultirotorMixer/CMakeFiles/mixer_gen_6dof.dir/build.make:82: recipe for target 'src/lib/mixer/MultirotorMixer/mixer_multirotor_6dof.generated.h' failed
make[4]: *** [src/lib/mixer/MultirotorMixer/mixer_multirotor_6dof.generated.h] Error 1
CMakeFiles/Makefile2:3737: recipe for target 'src/lib/mixer/MultirotorMixer/CMakeFiles/mixer_gen_6dof.dir/all' failed
make[3]: *** [src/lib/mixer/MultirotorMixer/CMakeFiles/mixer_gen_6dof.dir/all] Error 2
[  3%] Built target uorb_headers
CMakeFiles/Makefile2:29637: recipe for target 'platforms/posix/CMakeFiles/gazebo.dir/rule' failed
make[2]: *** [platforms/posix/CMakeFiles/gazebo.dir/rule] Error 2
Makefile:9117: recipe for target 'gazebo' failed
make[1]: *** [gazebo] Error 2
Makefile:217: recipe for target 'amovlab_sitl_default' failed
make: *** [amovlab_sitl_default] Error 2

wcm-123, 2021-11-24:
> 应该是环境还没有安装完全，请参考 https://github.com/amov-lab/Prometheus/wiki/%E4%BB%BF%E7%9C%9F%E7%8E%AF%E5%A2%83%E9%85%8D%E7%BD%AE#px4%E7%BC%96%E8%AF%91%E7%8E%AF%E5%A2%83%E9%85%8D%E7%BD%AE%E5%8F%8A%E5%9B%BA%E4%BB%B6%E4%BB%A3%E7%A0%81%E5%AE%89%E8%A3%85

我看到少了 numpy，然后我执行pip3 install --user numpy还是这样的错
CMakeFiles/Makefile2:29637: recipe for target 'platforms/posix/CMakeFiles/gazebo.dir/rule' failed
make[2]: *** [platforms/posix/CMakeFiles/gazebo.dir/rule] Error 2
Makefile:9117: recipe for target 'gazebo' failed
make[1]: *** [gazebo] Error 2
Makefile:217: recipe for target 'amovlab_sitl_default' failed
make: *** [amovlab_sitl_default] Error 2

473867143, 2021-11-25:
继续先运行下面步骤呗，执行 source ./ubuntu.sh 把PX4相关工具链都安装了，再去编译

wcm-123, 2021-11-25:
> 继续先运行下面步骤呗，执行 source ./ubuntu.sh 把PX4相关工具链都安装了，再去编译
在执行source ./ubuntu.sh
时，中间会出现  Could not find a version that satisfies the requirement argparse>=1.2 (from -r /home/miao1/prometheus_px4/Tools/setup/requirements.txt (line 2)) (from versions: )
No matching distribution found for argparse>=1.2 (from -r /home/miao1/prometheus_px4/Tools/setup/requirements.txt (line 2))
重复执行了几次source ./ubuntu.sh还是报上面的错误

473867143, 2021-12-01:
有尝试直接安装仿真镜像吗？

wcm-123, 2021-12-01:
没有，我是在双系中的ubuntu18.04下运行Prometheus

IndexMark, 2021-12-22:
报错信息往上翻翻看，可能是gstreamer1.0和gstreamer1.0-base没装（我就是这种情况）
按照wiki里给的解决方案

缺少gstreamer：

sudo apt-get install libgstreamer1.0-0 gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly gstreamer1.0-libav gstreamer1.0-doc gstreamer1.0-tools gstreamer1.0-x gstreamer1.0-alsa gstreamer1.0-gl gstreamer1.0-gtk3 gstreamer1.0-qt5 gstreamer1.0-pulseaudio

装完后再试试看

closed 2022-01-06 by 473867143