amov-lab/Prometheus #194: 数字识别出错

dorothy1122, 2021-12-01:
运行数字识别的roslaunch文件，发现**[digitnum_det-16]process has died [pid 27602, exit code 1,...**的问题，在python2.7/python3.6运行环境下都有问题，pytorch_mnist_det.py代码编译没问题，应该是torch的问题，但是已经安装了pytorch，想问一下在pytorch_mnist_det.py里面是怎么找torch？

dorothy1122, 2021-12-01:
[image]

dorothy1122, 2021-12-01:
[image]

jario-jin, 2021-12-02:
`No module named builtins`，应该是需要运行`pip install future`
如果想看`torch`有没有问题，需要再`terminal`里运行`python/python3`，然后在里面输出`import torch`如果没报错就没问题。

dorothy1122, 2021-12-02:
> 

torch没问题，环境是python3.6.13，torch版本是1.4.0+cpu，运行脚本还是有错误，麻烦你们看下是不是代码问题呢？

dorothy1122, 2021-12-02:
![2021-12-02 18-59-21 的屏幕截图](https://user-images.githubusercontent.com/65701107/144412439-72fd7004-ab70-4436-973b-d067d251241e.png
[image]
)

dorothy1122, 2021-12-02:
[image]

dorothy1122, 2021-12-02:
[image]

bzbsng, 2021-12-07:
我也有同样的问题，请问解决了吗

dorothy1122, 2021-12-08:
> 我也有同样的问题，请问解决了吗

解决了，px4的问题，安装px4的时候中间有一步我之前用sudo make去做的，装出来编译也没有问题，但是我换了一种安装的方式，再去运行这个就没有问题了

bzbsng, 2021-12-08:
> > 我也有同样的问题，请问解决了吗
> 
> 解决了，px4的问题，安装px4的时候中间有一步我之前用sudo make去做的，装出来编译也没有问题，但是我换了一种安装的方式，再去运行这个就没有问题了

我安装px4就是按照这个wiki里面给的步骤，你说的换了一种安装方式有没有参考网站啊，能不能发一下，感谢感谢!

closed 2021-12-16 by 473867143