amov-lab/Prometheus #196: 请问fastplanner为什么不起作用了？

Handsome-B, 2021-12-08:
编译没有任何问题，运行也没有报错，但在rviz中给目标点后并不进行规划，这是为什么？

473867143, 2021-12-08:
你可以尝试一下ego-planner 
https://github.com/amov-lab/Prometheus/wiki/Promehteus%E4%BB%BF%E7%9C%9F%E5%8A%9F%E8%83%BD#launch_ego_planner%E6%96%87%E4%BB%B6%E5%A4%B9%E6%B5%8B%E8%AF%95%E9%98%B6%E6%AE%B5

Handsome-B, 2021-12-08:
> 你可以尝试一下ego-planner [https://github.com/amov-lab/Prometheus/wiki/Promehteus%E4%BB%BF%E7%9C%9F%E5%8A%9F%E8%83%BD#launch_ego_planner %E6%96%87%E4%BB%B6%E5%A4%B9%E6%B5%8B%E8%AF%95%E9%98%B6%E6%AE%B5](https://github.com/amov-lab/Prometheus/wiki/Promehteus%E4%BB%BF%E7%9C%9F%E5%8A%9F%E8%83%BD#launch_ego_planner%E6%96%87%E4%BB%B6%E5%A4%B9%E6%B5%8B%E8%AF%95%E9%98%B6%E6%AE%B5)

感谢回复！我也尝试过egoplanner并且成功运行了，但我希望能够把程序给部署到我们刚买的p450实机里面（无D435i版本，现已自己装上d435i），我研究了好久egoplanner的所有launch文件，发现结构好复杂，顿时不知道要怎么下手了，相比之下fastplanner似乎要简单不少。请问可以提供fastplanner的成功运行方法或者egoplanner的实机部署方法吗？非常感谢！

473867143, 2021-12-16:
暂时没有哈，抱歉

closed 2021-12-16 by 473867143