amov-lab/Prometheus #206: 固件烧录

Sigmund-SJTUer, 2021-12-30:
请问如何将我们的固件代码烧录进PixHawk

Chentao2000, 2021-12-31:
点击QGC ，选择烧录固件，选择你要烧录固件即可

Sigmund-SJTUer, 2022-01-01:
> 点击QGC ，选择烧录固件，选择你要烧录固件即可

请问用的是哪一版的QGC，我用的是3.5.6，没有找到烧录本地固件代码的选项。默认是烧录官方PX4代码

Chentao2000, 2022-01-01:
> > 点击QGC ，选择烧录固件，选择你要烧录固件即可
> 
> 请问用的是哪一版的QGC，我用的是3.5.6，没有找到烧录本地固件代码的选项。默认是烧录官方PX4代码
点击高级选项 选择自定义固件

Chentao2000, 2022-01-01:
> > 点击QGC ，选择烧录固件，选择你要烧录固件即可
> 
> 请问用的是哪一版的QGC，我用的是3.5.6，没有找到烧录本地固件代码的选项。默认是烧录官方PX4代码

> 点击高级选项，选择自定义固件

Chentao2000, 2022-01-01:
[image]
[image]

Sigmund-SJTUer, 2022-01-01:
> [image] [image]
不知道为什么，我没有Firmware setup部分

Chentao2000, 2022-01-01:
. 有没有点击高级设置?

Chentao2000, 2022-01-01:
1. 有没有连接上PX4？
2.  有没有点击高级设置？

Sigmund-SJTUer, 2022-01-02:
> 请问应该烧录那个文件呢

473867143, 2022-01-06:
https://prometheus-wiki.readthedocs.io/zh_CN/latest/docs/baseconfig_for_px4/2-%E5%9B%BA%E4%BB%B6%E5%88%B7%E5%86%99.html
可以看看固件烧写部分，相应视频在b站