amov-lab/Prometheus #209: 室外自主降落问题

chengfeng911, 2022-01-05:
[image]
我在室外进行二维码降落，其中的参数已经修改过了，多次在开机之后第一次在正常启动roslaunch p450_experiment p450_gps_onboard.launch，此时的usb相机还没有检测到二维码，在终端显示的x、y轴显示数字能有三四米的数据，而z轴的数据是正常的误差变化不大。而且在所有成功运行并且切到offboard模式进行降落时候，降落到三四十厘米的时候，会重新上升，重复几次然后相机就会丢失二维码画面。相机的方向、机头的方向、二维码都是朝东的，每次飞机都会飞到以二维码为中心的上方稍微偏左的位置，然后usb相机丢失二维码画面。紧接着就彻底丢失二维码了，降落不下来了

473867143, 2022-01-05:
https://github.com/amov-lab/Prometheus/blob/7a4d5f3d78495dfeb7129c054b9a3fcf1ef818db/Modules/mission/autonomous_landing/autonomous_landing_outdoor.cpp?_pjax=%23js-repo-pjax-container%2C%20div%5Bitemtype%3D%22http%3A%2F%2Fschema.org%2FSoftwareSourceCode%22%5D%20main%2C%20%5Bdata-pjax-container%5D#L293

降落处理逻辑需要修改一下，因为室外gps数据情况不一样，会导致重复降落情况

chengfeng911, 2022-01-06:
> 

我们使用是阿木自己的p450无人机，p450自己本身的室外降落代码，我看的代码上面给的备注信息是已经做了需改，和室内降落代码逻辑不太一样。室外使用gps一次降落一次成功都没有，如果是还需要修改的话，能否给一点修改建议，尽可能减少重复降落的情况

chengfeng911, 2022-01-06:
> https://github.com/amov-lab/Prometheus/blob/7a4d5f3d78495dfeb7129c054b9a3fcf1ef818db/Modules/mission/autonomous_landing/autonomous_landing_outdoor.cpp?_pjax=%23js-repo-pjax-container%2C%20div%5Bitemty [...]

我们使用是阿木自己的p450无人机，p450自己本身的室外降落代码，我看的代码上面给的备注信息是已经做了需改，和室内降落代码逻辑不太一样。室外使用gps一次降落一次成功都没有，如果是还需要修改的话，能否给一点修改建议，尽可能减少重复降落的情况

473867143, 2022-01-06:
https://github.com/amov-lab/Prometheus/blob/7a4d5f3d78495dfeb7129c054b9a3fcf1ef818db/Modules/control/src/px4_sender.cpp?_pjax=%23js-repo-pjax-container%2C%20div%5Bitemtype%3D%22http%3A%2F%2Fschema.org%2FSoftwareSourceCode%22%5D%20main%2C%20%5Bdata-pjax-container%5D#L313

在px4_sender中使用第一种降落模式，然后注释掉高于Disarmed_height高度处理情况。这样子一旦高度降落下来，就直接进入飞控中的land模式，但是这样可能精度不是特别好。

之所以会上升，是因为定位精度不好，室外数据有误差，随时间误差累积越来越大，所以一般都建议起飞前reboot飞控，重置ekf。

chengfeng911, 2022-01-07:
> 你好，你这边说的使用第一种降落模式指的是land_mode==1的情况，还是land_mode==0的时候

473867143, 2022-01-07:
// 【Land】 降落。两种降落方式： 只有加载了参数Land_mode为1时，启用第二种降落方式；默认启用第一种降落方式。
        //  第一种：当前位置原地降落，降落后会自动上锁，且切换为mannual模式
        //  第二种：当前位置原地降落，降落中到达Disarm_height后，切换为飞控中land模式

473867143, 2022-01-07:
上面说的有误，建议修改使用第二种降落模式，land_mode==1

chengfeng911, 2022-01-07:
好的，谢谢您的回答，还有个问题，想咨询一下您，我使用的降落代码是autonomous_landing_outdoor.cpp，按照代码逻辑，当检测目标丢失时，应该是会进入HOLD模式悬停，可以实验时，每次检测不到时，飞机就会上升，是gps定位不准的缘故吗，毕竟这边px4——sender.cpp这边接受到hold命令时，也是使用了_Drone_State.position的

473867143, 2022-01-07:
https://github.com/amov-lab/Prometheus/blob/7a4d5f3d78495dfeb7129c054b9a3fcf1ef818db/Modules/mission/autonomous_landing/autonomous_landing.cpp?_pjax=%23js-repo-pjax-container%2C%20div%5Bitemtype%3D%22http%3A%2F%2Fschema.org%2FSoftwareSourceCode%22%5D%20main%2C%20%5Bdata-pjax-container%5D#L431

先是短暂悬停，10s后就当前位置上升，搜寻二维码。我建议你在你当前环境中，先试试指点飞行的各个接口，什么位置控制，速度控制，hold指令等等。因为降落也是于指点飞行控制一样的