amov-lab/Prometheus #212: 仿真运行问题

Zxr1230, 2022-01-12:
在跑roslaunch prometheus_gazebo sitl.launch 时出现了下面的问题，求教，谢谢！
[image]

RLException: Roslaunch got a 'No such file or directory' error while attempting to run:

gnome-terminal --tab -- /home/xqz5222/Prometheus/devel/lib/prometheus_station/ground_station __name:=ground_station __log:=/home/xqz5222/.ros/log/bc30613e-73bc-11ec-8e95-00155d71c412/ground_station-14.log

Please make sure that all the executables in this command exist and have
executable permission. This is often caused by a bad launch-prefix.
The traceback for the exception was written to the log file

Junhuan-Peng, 2022-02-28:
没有gnome-terminal。试一试`sudo apt install gnome-terminal`。我在WSL下面有同样的问题，解决了，不知道你那边行不行

Zxr1230, 2022-02-28:
> 
谢谢！我用的也是WSL，这个问题解决啦！不过为什么gazebo里什么也没有呢？
[image]

Zxr1230, 2022-02-28:
> 没有gnome-terminal。试一试`sudo apt install gnome-terminal`。我在WSL下面有同样的问题，解决了，不知道你那边行不行
谢谢！我用的也是WSL，这个问题解决啦！不过为什么gazebo里什么也没有呢？
[image]

Junhuan-Peng, 2022-02-28:
我执行命令之后就没有问题了～你看看你是不是模型的路径加载得不太对。看看终端里的日志有没有提示相关的信息。我这边是自己用conda创建了虚拟环境装的Python的包，如果忘记激活虚拟化也出现过场景没有模型的问题。<html>
<head>
    <meta http-equiv="Content-Type" content="text/html; charset=UTF-8">
</head>
<body>
<br/><br/><br/><div class="ntes-mailmaster-quote" style="padding-top: 1px; padding-bottom: 1px" >
    <div style=" margin-top: 2em; margin-bottom: 1em; font-size: 14px; line-height: 1.25; color: #78787a; " >---- 回复的原邮件 ----</div>
    <div style=" margin-bottom: 1em; font-size: 12px; line-height: 1.25; color: #232324; padding: 0.5em 0.25em; border-radius: 0.67em; background-color: #f0f0f0; " >
        <table width="100%" cellpadding="0" cellspacing="9" border="0">
            <tr>
                <td valign="top" style=" width: 4.25em; font-size: 12px; line-height: 1.25; color: #78787a; " >
                    发件人
                </td>
                <td valign="top" style=" font-size: 12px; line-height: 1.25; color: #232324; word-break: break-all; " >
                    <a class="mail-from" style="text-decoration:none;color:#0886e8;" ***@***.******@***.***&gt;</a>
                </td>
            </tr>
            <tr>
                <td valign="top" style=" width: 4.25em; font-size: 12px; line-height: 1.25; color: #78787a; " >
                    日期
                </td>
                <td class="mail-date" valign="top" style=" font-size: 12px; line-height: 1.25; color: #232324; word-break: break-all; " >
                    2022年02月28日 23:00
                </td>
            </tr>
            <tr style="">
                <td valign="top" style=" width: 4.25em; font-size: 12px; line-height: 1.25; color: #78787a; " >
                    收件人
                </td>
                <td valign="top" style=" font-size: 12px; line-height: 1.25; color: #232324; word-break: break-all; " >
                    <a class="mail-to" style="text-decoration:none;color:#0886e8;" ***@***.******@***.***&gt;</a>
                </td>
            </tr>
            <tr style="">
                <td valign="top" style=" width: 4.25em; font-size: 12px; line-height: 1.25; color: #78787a; " >
                    抄送至
                </td>
                <td valign="top" style=" font-size: 12px; line-height: 1.25; color: #232324; word-break: break-all; " >
                    <a class="mail-cc" style="text-decoration:none;color:#0886e8;" ***@***.******@***.***&gt;</a>、<a class="mail-cc" style="text-decoration:none;color:#0886e8;" ***@***.******@***.***&gt;</a>
                </td>
            </tr>
            <tr>
                <td valign="top" style=" width: 4.25em; font-size: 12px; line-height: 1.25; color: #78787a; " >
                    主题
                </td>
                <td class="mail-subject" valign="top" style=" font-size: 12px; line-height: 1.25; color: #232324; word-break: break-all; " >
                    Re: [amov-lab/Prometheus] 仿真运行问题 (Issue #212)
                </td>
            </tr>
        </table>
    </div>
    <div><p></p>
<blockquote>
</blockquote>
<p dir="auto">谢谢！我用的也是WSL，这个问题解决啦！不过为什么gazebo里什么也没有呢？<br>
[image]</p>

<p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">&mdash;<br />
<script type="application/ld+json">[
{
"target": "https://github.com/amov-lab/Prometheus/issues/212#issuecomment-1054342731",
"url": "https://github.com/amov-lab/Prometheus/issues/212#issuecomment-1054342731",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script></div>
</div>
</body>
</html>

Junhuan-Peng, 2022-03-01:
我执行命令之后就没有问题了～你看看你是不是模型的路径加载得不太对。看看终端里的日志有没有提示相关的信息。我这边是自己用conda创建了虚拟环境装的Python的包，如果忘记激活虚拟化也出现过场景没有模型的问题。