amov-lab/Prometheus #216: ./compile_all.sh编译错误

Zxr1230, 2022-01-14:
在./compile_all.sh编译过程中有如下几个错误，不知道是否有影响，有影响的话该如何解决？谢谢！
错误1：
[image]
错误2：
[image]
错误3：
[image]
错误4：
[image]

473867143, 2022-01-15:
感觉不影响，有终端编译终止吗？

Zxr1230, 2022-01-15:
> 

谢谢，没有终止，继续往下跑了，这是不是意味着没有影响？

Zxr1230, 2022-01-15:
> 感觉不影响，有终端编译终止吗？

谢谢，没有终止，继续往下跑了，这是不是意味着没有影响？

473867143, 2022-01-16:
是的

closed 2022-01-16 by 473867143