amov-lab/Prometheus #219: A* 算法路径问题

wcm-123, 2022-01-20:
我将sitl_astar_3dlidar.launch文件中的障碍物膨胀距离设置成  0，如下
      
        <param name="map/inflate" value="0" type="double"/>

然后把global_planner.cpp文件中的安全距离设为0，如下
    // 安全距离，若膨胀距离设置已考虑安全距离，建议此处设为0
    nh.param("global_planner/safe_distance", safe_distance, 0); 

再把地图的膨胀距离也设置成0，如下
    // 地图膨胀距离，单位：米
    nh.param("map/inflate", inflate_,  0);
改完之后进行的路径规划结果是，路径依然离障碍物很远，按道理会规划出一条比较靠近障碍物的路径，但是集中怎么规划都是与障碍物很远，有知道的小伙伴吗

473867143, 2022-01-20:
1、不是在代码中修改膨胀距离，在launch文件中哈，sitl_astar_3dlidar.launch。

2、3d雷达不一定能够正常跑起来，可以运行2d的

wcm-123, 2022-01-20:
> 

无人机是先规划出路径，然后再飞，我只需要规划出路径即可，传统A*算法就是这样的，有一个问题没有回答，就是膨胀距离设置为0后，为啥规划出的路径离障碍物还是这么远

wcm-123, 2022-01-20:
> 

路径规划中，生成的轨迹在无人机运动过程中会消失，不能在八叉树地图上显示，那么我没法截取路径规划的图

473867143, 2022-01-20:
> 

膨胀距离修改感觉没有生效吧，在launch文件参数设置中生效