amov-lab/Prometheus #222: 关于路径规划和避障问题——节点无法通讯上

four1fire, 2022-02-05:
在做路径规划算法时，我使用Promethues打开了gazebo和rviz后，始终无法设置目标点，无论是用3d nav设置还是在终端输入节点。飞机起飞以后就不动，无法接收到目标点，规划出路径。终端的报错如下所示：
ERROR: cannot launch node of type [prometheus_local_planning/local_planner_main]: Cannot locate node of type [local_planner_main] in package [prometheus_local_planning]. Make sure file exists in package path and permission is set to executable (chmod +x)
[我检查了包是有这个节点的，而且将包中的py文件都设置了一遍chmod+x，但还是有这个毛病，大佬能不能帮忙解答一下。
[image]
[image]
[image]

kirito-biter, 2022-03-26:
我也用不了

kirito-biter, 2022-03-26:
好像这个停止维护了？