amov-lab/Prometheus #228: ./compile_control.sh出错

bzbsng, 2022-03-07:
compilation terminated.
make[1]: *** [CMakeFiles/Makefile2:237：CMakeFiles/ugv_estimator.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
make[2]: *** [CMakeFiles/fake_vicon.dir/build.make:63：CMakeFiles/fake_vicon.dir/src/fake_vicon.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:200：CMakeFiles/fake_vicon.dir/all] 错误 2
/home/leonard/Prometheus/Modules/ugv_control/src/ugv_terminal_control.cpp:3:10: fatal error: Eigen/Eigen: 没有那个文件或目录
 #include <Eigen/Eigen>
          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ugv_terminal_control.dir/build.make:63：CMakeFiles/ugv_terminal_control.dir/src/ugv_terminal_control.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:2157：CMakeFiles/ugv_terminal_control.dir/all] 错误 2
In file included from /home/leonard/Prometheus/Modules/ugv_control/include/ugv_controller.h:11:0,
                 from /home/leonard/Prometheus/Modules/ugv_control/src/ugv_controller.cpp:1:
/home/leonard/Prometheus/Modules/ugv_control/include/math_utils.h:15:10: fatal error: Eigen/Eigen: 没有那个文件或目录
 #include <Eigen/Eigen>
          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ugv_controller.dir/build.make:63：CMakeFiles/ugv_controller.dir/src/ugv_controller.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:275：CMakeFiles/ugv_controller.dir/all] 错误 2
make: *** [Makefile:141：all] 错误 2
Invoking "make -j16 -l16" failed

这种情况怎么解决啊

473867143, 2022-03-21:
https://github.com/amov-lab/Prometheus/blob/d5df6564523fc61b436e95b671f3a3bd18cb7407/compile_control.sh#L4

注释掉，重新编译

zhaoguoming, 2022-05-12:
我也遇到了一样的问题，怎么解决的？

zhaoguoming, 2022-05-12:
Base path: /home/zgm/catkin_promethues/src/Prometheus
Source space: /home/zgm/catkin_promethues/src/Prometheus/Modules/ground_station
Build space: /home/zgm/catkin_promethues/src/Prometheus/build/ground_station
Devel space: /home/zgm/catkin_promethues/src/Prometheus/devel
Install space: /home/zgm/catkin_promethues/src/Prometheus/install
####
#### Running command: "make cmake_check_build_system" in "/home/zgm/catkin_promethues/src/Prometheus/build/ground_station"
####
####
#### Running command: "make -j16 -l16" in "/home/zgm/catkin_promethues/src/Prometheus/build/ground_station"
####
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target nav_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target nav_msgs_generate_messages_py
[  0%] Built target nav_msgs_generate_messages_lisp
[  0%] Built target nav_msgs_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target prometheus_station_generate_messages_lisp
[  0%] Built target prometheus_station_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target nav_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_nodejs
[ 14%] Generating EusLisp manifest code for prometheus_station
[ 14%] Built target prometheus_station_generate_messages_py
[ 14%] Built target prometheus_station_gencpp
[ 14%] Built target prometheus_station_generate_messages_nodejs
[ 42%] Built target ground_station_msg
[ 71%] Built target ground_station_aruco
[100%] Built target ground_station
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/geneus/geneus_main.py", line 137, in genmain
    pkg_map = get_pkg_map()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/geneus/geneus_main.py", line 56, in get_pkg_map
    pkgs = packages.find_packages(ws)
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/packages.py", line 96, in find_packages
    raise RuntimeError('\n'.join(duplicates))
RuntimeError: Multiple packages found with the same name "bspline_opt":
- Prometheus/Modules/ego_planner/ego-planner-swarm/src/planner/bspline_opt
- Prometheus/Modules/planning/FastPlanner/bspline_opt
Multiple packages found with the same name "path_searching":
- Prometheus/Modules/ego_planner/ego-planner-swarm/src/planner/path_searching
- Prometheus/Modules/planning/FastPlanner/path_searching
Multiple packages found with the same name "plan_env":
- Prometheus/Modules/ego_planner/ego-planner-swarm/src/planner/plan_env
- Prometheus/Modules/planning/FastPlanner/plan_env
Multiple packages found with the same name "rviz_plugins":
- Prometheus/Modules/ego_planner/ego-planner-swarm/src/uav_simulator/Utils/rviz_plugins
- Prometheus/Modules/planning/Utils/rviz_plugins
Multiple packages found with the same name "traj_utils":
- Prometheus/Modules/ego_planner/ego-planner-swarm/src/planner/traj_utils
- Prometheus/Modules/planning/FastPlanner/traj_utils
ERROR:  Multiple packages found with the same name "bspline_opt":
- Prometheus/Modules/ego_planner/ego-planner-swarm/src/planner/bspline_opt
- Prometheus/Modules/planning/FastPlanner/bspline_opt
Multiple packages found with the same name "path_searching":
- Prometheus/Modules/ego_planner/ego-planner-swarm/src/planner/path_searching
- Prometheus/Modules/planning/FastPlanner/path_searching
Multiple packages found with the same name "plan_env":
- Prometheus/Modules/ego_planner/ego-planner-swarm/src/planner/plan_env
- Prometheus/Modules/planning/FastPlanner/plan_env
Multiple packages found with the same name "rviz_plugins":
- Prometheus/Modules/ego_planner/ego-planner-swarm/src/uav_simulator/Utils/rviz_plugins
- Prometheus/Modules/planning/Utils/rviz_plugins
Multiple packages found with the same name "traj_utils":
- Prometheus/Modules/ego_planner/ego-planner-swarm/src/planner/traj_utils
- Prometheus/Modules/planning/FastPlanner/traj_utils
CMakeFiles/prometheus_station_generate_messages_eus.dir/build.make:60: recipe for target '/home/zgm/catkin_promethues/src/Prometheus/devel/share/roseus/ros/prometheus_station/manifest.l' failed
make[2]: *** [/home/zgm/catkin_promethues/src/Prometheus/devel/share/roseus/ros/prometheus_station/manifest.l] Error 3
CMakeFiles/Makefile2:824: recipe for target 'CMakeFiles/prometheus_station_generate_messages_eus.dir/all' failed
make[1]: *** [CMakeFiles/prometheus_station_generate_messages_eus.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j16 -l16" failed