amov-lab/Prometheus #229: 自主降落，通过程序切offboard问题

lllz815, 2022-03-07:
您好，请教一个室外自主降落的问题，如何在空中通过代码切offboard，而飞机不会坠落，整个流程是：先position模式通过遥控器起飞到5米，在此时发送一个指令切offboard，然后通过代码切到offboard,飞机再执行后面的二维码自主降落程序
	Command_Now.header.stamp = ros::Time::now();
        Command_Now.Mode  = prometheus_msgs::ControlCommand::Idle;
        Command_Now.Command_ID = Command_Now.Command_ID + 1;
        Command_Now.source = NODE_NAME;
        Command_Now.Reference_State.yaw_ref = 999;
        command_pub.publish(Command_Now);   
        cout << "Switch to OFFBOARD and arm ..."<<endl;
        ros::Duration(5.0).sleep();
        我使用的是这段代码去切offboard，虽然能够切到offboard模式，但是此时飞机会坠落。
如何保证飞机不坠落，又能跑通整个流程呢

473867143, 2022-03-21:
这个不能哦，这个是解锁切换offboard，且该模式处于怠速模式，给无人机的控制指令全为0,0,0

如果要在position下切换offboard，你发送指令之后直接切换offboard，不要使用999这个操作，这个指令不只是解锁。另外一种，你先使用999解锁并且换offboard，然后在发送指令，这个样也可以。

具体可以参考自主降落demo，    https://github.com/amov-lab/Prometheus/blob/d5df6564523fc61b436e95b671f3a3bd18cb7407/Modules/mission/autonomous_landing/autonomous_landing.cpp#L224

lllz815, 2022-03-21:
那不使用999的指令的话，怎么切offboard