amov-lab/Prometheus #231: 编译 planning 遇到的问题

6wa-car, 2022-03-10:
Running command: "cmake /home/model-aircraft/Prometheus/Modules/planning -DCATKIN_DEVEL_PREFIX=/home/model-aircraft/Prometheus/devel -DCMAKE_INSTALL_PREFIX=/home/model-aircraft/Prometheus/install -G Unix Makefiles" in "/home/model-aircraft/Prometheus/build/planning"
####
-- Using CATKIN_DEVEL_PREFIX: /home/model-aircraft/Prometheus/devel
-- Using CMAKE_PREFIX_PATH: /home/model-aircraft/car_ws/devel;/home/model-aircraft/catkin_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/model-aircraft/car_ws/devel;/home/model-aircraft/catkin_ws/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/model-aircraft/Prometheus/build/planning/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 13 packages in topological order:
-- ~~  - prometheus_cmake_utils
-- ~~  - prometheus_planning_sim
-- ~~  - prometheus_pose_utils
-- ~~  - prometheus_local_planning
-- ~~  - plan_env
-- ~~  - bspline_opt
-- ~~  - path_searching
-- ~~  - prometheus_global_planning
-- ~~  - prometheus_hybrid_astar
-- ~~  - prometheus_swarm_planning
-- ~~  - traj_utils
-- ~~  - prometheus_plan_manage
-- ~~  - rviz_plugins
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'prometheus_cmake_utils'
-- ==> add_subdirectory(Utils/cmake_utils)
-- +++ processing catkin package: 'prometheus_planning_sim'
-- ==> add_subdirectory(planning_simulator)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   date_time
--   iostreams
--   serialization
--   chrono
--   atomic
--   regex
-- Could NOT find ensenso (missing: ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
** WARNING ** io features related to ensenso will be disabled
-- Could NOT find DAVIDSDK (missing: DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** io features related to davidSDK will be disabled
-- Could NOT find DSSDK (missing: _DSSDK_LIBRARIES) 
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

** WARNING ** io features related to libusb-1.0 will be disabled
-- Could NOT find ensenso (missing: ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
** WARNING ** visualization features related to ensenso will be disabled
-- Could NOT find DAVIDSDK (missing: DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** visualization features related to davidSDK will be disabled
-- Could NOT find DSSDK (missing: _DSSDK_LIBRARIES) 
** WARNING ** visualization features related to dssdk will be disabled
-- Could NOT find RSSDK (missing: _RSSDK_LIBRARIES) 
** WARNING ** visualization features related to rssdk will be disabled

Ubuntu 18.04 原生系统和Prometheus镜像 在执行  ./compile_all.sh 到planning部分都会遇到这个问题
有没有大佬能给解答一下

Chentao2000, 2022-03-27:
貌似是你系统组件的问题：
从这一句看：
-- The imported target "vtkRenderingPythonTkWidgets" references the file
"/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist. Possible reasons include:

[image]
```
sudo ln -s /usr/lib/python2.7/dist-packages/vtk/libvtkRenderingPythonTkWidgets.x86_64-linux-gnu.so /usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so

sudo ln -s /usr/bin/vtk6 /usr/bin/vtk 
```

Chentao2000, 2022-03-27:
接上一条： 参考链接：

https://blog.csdn.net/weixin_45617478/article/details/104513572