amov-lab/Prometheus #243: 编译 Prometheus时运行./compile_all.sh报错

HZXTYT, 2022-03-28:
#### Running command: "make -j8 -l8" in "/home/wrj/Prometheus/build/ego_planner"
####
[  0%] Built target nav_msgs_generate_messages_cpp
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_LQRTrajectory
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_Odometry
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_Serial
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_TRPYCommand
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_Gains
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_PPROutputData
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_OutputData
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_PositionCommand
[  0%] Built target nav_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_Corrections
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_StatusData
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_AuxCommand
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_SO3Command
[  0%] Built target nav_msgs_generate_messages_lisp
[  0%] Built target _quadrotor_msgs_generate_messages_check_deps_PolynomialTrajectory
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target nav_msgs_generate_messages_eus
[  0%] Built target nav_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_nodejs
make[2]: *** 没有规则可制作目标“/usr/lib/x86_64-linux-gnu/libGL.so”，由“/home/wrj/Prometheus/devel/lib/map_generator/random_forest_new” 需求。 停止。
CMakeFiles/Makefile2:2512: recipe for target 'ego-planner-swarm/src/uav_simulator/map_generator/CMakeFiles/random_forest_new.dir/all' failed
make[1]: *** [ego-planner-swarm/src/uav_simulator/map_generator/CMakeFiles/random_forest_new.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
make[2]: *** 没有规则可制作目标“/usr/lib/x86_64-linux-gnu/libGL.so”，由“/home/wrj/Prometheus/devel/lib/map_generator/random_forest” 需求。 停止。
CMakeFiles/Makefile2:2741: recipe for target 'ego-planner-swarm/src/uav_simulator/map_generator/CMakeFiles/random_forest.dir/all' failed
make[1]: *** [ego-planner-swarm/src/uav_simulator/map_generator/CMakeFiles/random_forest.dir/all] Error 2
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  1%] Built target pose_utils
[  1%] Built target std_msgs_generate_messages_py
make[2]: *** 没有规则可制作目标“/usr/lib/x86_64-linux-gnu/libGL.so”，由“/home/wrj/Prometheus/devel/lib/libplan_env.so” 需求。 停止。
CMakeFiles/Makefile2:3032: recipe for target 'ego-planner-swarm/src/planner/plan_env/CMakeFiles/plan_env.dir/all' failed
make[1]: *** [ego-planner-swarm/src/planner/plan_env/CMakeFiles/plan_env.dir/all] Error 2
[  2%] Built target obj_generator
[  2%] Built target std_msgs_generate_messages_nodejs
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
Base path: /home/wrj/Prometheus
Source space: /home/wrj/Prometheus/Modules/swarm_control
Build space: /home/wrj/Prometheus/build/swarm_control
Devel space: /home/wrj/Prometheus/devel
Install space: /home/wrj/Prometheus/install
####

Chentao2000, 2022-03-29:
报错来源：
```
make[2]: *** 没有规则可制作目标“/usr/lib/x86_64-linux-gnu/libGL.so”
```
解决方法
```
// 查找libGL.so的位置
sudo find / -iname "*libGL.so*"
// 查看链接状态
usr/lib/x86_64-linux-gnu$ ls -l libGL.so
// 建立链接
sudo ln -s /usr/lib/x86_64-linux-gnu/libGL.so.1.0.0 /usr/lib/x86_64-linux-gnu/libGL.so
```

HZXTYT, 2022-03-31:
我在CSDN上找到过这种方法，但并没有起作用，最终没办法，重装系统再部署仿真平台，这次就一次编译通过了。

> 报错来源：
> 
> ```
> make[2]: *** 没有规则可制作目标“/usr/lib/x86_64-linux-gnu/libGL.so”
> ```
> 
> 解决方法
> 
> ```
> // 查找libGL.so的位置
> sudo find / -iname "*libGL.so*"
> // 查看链接状态
> usr/lib/x86_64-linux-gnu$ ls -l libGL.so
> // 建立链接
> sudo ln -s /usr/lib/x86_64-linux-gnu/libGL.so.1.0.0 /usr/lib/x86_64-linux-gnu/libGL.so
> ```

我在CSDN上找到过这种方法，但并没有起作用，最终没办法，重装系统再部署仿真平台，这次就一次编译通过了。